AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #866: After every reboot I got this message: "AlDente wants to make changes. Enter your password to allow this."

taylor325, 2023-01-05:
Hello,
After every reboot I got this message: "AlDente wants to make changes. Enter your password to allow this."
I enter the password but AlDente not run. Launch at Login enabled.
I tried to remove it completely and install in earlier versions but I have a same bug.

M1 MacBook Pro, Ventura 13.1
AlDente Version: 1.20 (Free)
[image]

taylor325, 2023-01-08:
Still have this problem. AlDente 1.20 not run on background!
M1 MacBook Pro, Ventura 13.1
AlDente Version: 1.20 (Free)

papas68, 2023-01-09:
Still have this problem. AlDente pRO 1.20 not run on background
M1 MacBook Pro, (A2338),Ventura 13.1 
AlDente Pro Version: 1.20

MatthiasKerbl, 2023-01-12:
Hi @taylor325,

Sorry for the inconvenience.
Did you follow our uninstallation guide on our blog here?

Cheers,
Matthias

papas68, 2023-01-12:
YES and i had the same result

| George ***@***.***  ....... |

papas68, 2023-01-12:
I will do it now and updated

| George ***@***.***  ....... |

 
 

Yes I uninstalled a few times but still I have this issue :(

papas68, 2023-01-12:
I followed the instructions you told me but the result is the same
After every reboot I got this message: "AlDente wants to make changes. Enter your password to allow this."
I enter the password but AlDente not run. Launch at Login enabled.
How to uninstall AlDente Pro – AppHouseKitchen

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
How to uninstall AlDente Pro – AppHouseKitchen

 |

 |

 |

| George ***@***.***  ....... |

Attempt3035, 2023-01-16:
Hi,
I have just purchased Aldente Pro Perpetual Licence and I am getting the same issue! Is there any eta on when this could be fixed or is there any way I can help track down the reason for it? I too have tried uninstall, reinstall etc. I have included the contents of a debug file. (aldentepro-debug-2023-01-17-08/33/18.pasta)
`uxjw7lZ7TfAC1PDnWjCx8HQauurfKIMuwaRD5vVprfxlsX9Qm+6EMmyQQXDv9NpmMRTCny/5YgPCfCXilJcPt6w/2UdMIDi2NS4VKfVXs9fWFqLLmwzGW2fT2kmErO2Mb66rtlGEY/Yh0Sn6kURWwPX4IvSQS465S9AmKWpXCyxRARTySV92JI8gbVnV1v6rRf19i7W0cAxmUZv9CKVo8Ixc1INttRKdNrNpoUZTvStoWOQRAADM79mC7CqiRf0AoLha2LYqt0oEQgZBBBUVN/jL1mQbNxSxNLNTtxt8P3IzE6om7O7zllX0d+FAGJTNBNEbW8RxIpqZNS5E2IxJZpG341kJAAA9+JoF0msElQ2/RZzSclzGp/8rzuJC+CDXAzaySPsPA0YvYXMSoIa/HuXNmFJyJcTiFgTiyJZRAYj9QH7teInbnrXuAxTrO/Z5BFd7ti+TbtHzun2R4EygWrek4PNXvRNtStFZ5Ga1nEVsCzNza/2y298MvWCwYi6C9yga359qsAHEDqUcab9wucPWqQYAWE0nVu9T0pt4JW+QbuAp/DLskrlLIauys5MlIak5XA8VtZ29kM8o4ZxTXWXv6Y/oC//q5nHwgHpmMgCeoaaULcXv/Wvt3Ux1mEqMoqzqxc4eWh2kx8WRokzxrwPERxP3fJv6ajcAtTY9LuWTFOCa3+dhuV+ppa2YAwd1jvCtK6T9kVNaFYk3qUJdi7SDul5+DJnDEeRSjaa8BBaXbAXgrOFMuBF/QbgVIOnhoE8AVbUJF3AG/+cFymQzMXvUbvfCEKw9NGGZurg+LUkyk5uBUQRrfsuCo27VHIf75lJpqAHMXJddx77tKqkA9sPihWkaDCf3phoCjNRQQtdBemR2c7WyB6Z9ZDgHqZSv8JSUwVl0XNdHMTBXgccLSw5TMF14+l6wizACCVmDca/DjTk36TwwjxSpG/+B8nJaBMb2XFKcq/kWZuKqO/mNFbuj646NOWUFH5jkrYG7bQSkVhjroZa6NHseklZARqvwcIh0oek8m2MwpQwPCGk9WAfiZ57Ir2REEFR/Yv8ZVDyQWXA/T7jpP7/NQklTTPe1SBY1XKcXiRn/luLCZamgQGYU0qaCzynrG/wUy0yOqUkX1xNxav1woXpu7Iyyjdbifx8siHEMzpOd5U8kWfdmTHPTgCsO4nKXfJsbowlI+ensUlioKBOk1x6B1XUMBbzKO95tVctGT3RmmesmG1/lYPzQ981AnoGgbgntmWXWUm74xxmLl/A4yo/XPZ4ZdoK3CUE1o6jq2pJvJ833NkWvZwCY9xVWN4FypMfHo4KEHQZV0EH7fdeTkgbevducOtG+LkzSpdF7MO5RvQ7wdljNMWuJKd/mK3QTQHWZmBu7MZlTGUeGCmn5K275Xu92VOLitKb2wGTex3La4uXxGifGMkfTJvRv0mzRdNm/v8kubzx25Gs322+awzPRplyJ1toHQMxeSHDJMLS0PqBoOxPAc4R00KHk5+fxfozOnBjg6ou0j1qVLrmjThB23zLJHquLsRtwfUSBARkNQ5PJMcfb2RaF0pk42jJek+WFGuhVS+Q164bwBP2lNJBWSKgV587yj4pGqJMDTJF6++yxIOpOh4k+eoxFZcsaZGK4Eoz97K9GA74Dd+eFjkTO3C31rMLC6PToQNcqfrV4qAu1dEUYPTpXiaAsQ3XhwgLftk8kYDO/dqOILWKMJIyJ90OY/aA74tVpbwbYzMqgpPIPBI48rUfjB0lvORLGUjhiHMa4SLwrNVvC//LlTClwH3FMD09GkmgR3ge8cvPTWZACU1z2POjHLMh0ALcJTovEkaDfT/d6UC3I4Q6kzZjQPBE0yN8CCR1ha1qeej4seuOQ7nbET3cxOUuzhcw6NnnGN/715cDgiWJLWGjnBYMg1wNW5GJjlqEK2o2JRzNKRJ0yK/is85YH+wLDPnHY1SyBESjqPUbUG02ehamNyaj8HCuqIn9pb86lK9HhdUZ6gQcRB1g5C/mkta62boGBtnrXAbKUbLmCg/oAIA5ZSa5vtrK1S5XgKIgA1OuaLfLL0aBJza2L4lhEK/OobPFOLPCXYS1f39iYotV2FNCuryJzNproqfAXzs6reK7/KKZ5GbvPziWUgrwaI95wHg7fGiGeCklxWW0HUJaU8vtEad6YAO1yngYHhqyW6LKGdUHwBx79Pkt1yA0iLvOFqj4I8jLyWlOC2l3aaAl/UQXbQirUOEmVIp6p0IAqdmkVMBoKhqBqzUUHFTJBkAAYp1J9jpSPzZ4q85FvbXcfOqeGlH1exgualf1SYb+FyIU7r/lkouiowweYjnqFedJ6stway68Y4kyVcLH00tZYVGEB9vlF4arjKm2srInVKUohZkmiR5hDiRHUvKlR4C6ktVx5pzOx3n22tX9Jt3PrUa8HetOxfyb/nyoc7az8GEA74gfq+VL0eotnD/jMIq5HAX97GWjpan8Fvx3Zva+xz1JV8Elj/4ddDHOcgmR297zRQrlGPdCvEocC4TKzAiM/orghHfFnn2FNhWuNvbMgDiw+nqU8mvNqP+ulSXDhbGx6RhBbWHd7MPpOz/wRJGFeYKlsk2fWm5/n92aeu78fJxZ+3jwgxtMPwywE238Dx2cZe3M0/Jk6ofRk0wMTXK0ZGreFP0o8DwhnayBZ+dq7GHv1IqzhPiRrOZO1NLZr14Bh97LFLKBXz0sCDIV6Kh1tfdALEiAUbrFF+ZXRj51SpJH0Q+8UOUCGQ1+GSKq4tRNp3c9b80T+VMl/IhV/F9fnkz/DMy6wh6gMjQ+6ZgBfw/LXb+pI0tdIKADsi2PSmH4sybuR2Ak5I/aa8/X03lVeN2qjPSAn448I1O0faxXR6GlDiR17juLT6iECGFIhUednVaX+5v1rU7dhr2BTwNZG9KALbYa1NuV7NpYJ5sGiQlaSaVl1u6H4Mywpu46UqMQt9b3/OOEIuTCPQFVGFP86i8rImJQ30l3f3PplNQtGx7KESEawXAAgMn2I/RgbHpN0KL3V4GtIvRY9mHAFjOxDNYT2v56tjoWCdOPbFv+xppb5mD5WXJHvGtpyCnDVM8SaRMtyu/lIMtaImC4KLDcc7UJINT8v5+DB95FJsDtq5eZ37a8kUEoA1H1xM05PMdBoy76tRUkBu3jcirdP4r6HXnPuHqvHklgfYx08Vz90QaUouYGQcSHzPi8mYZI+y4B7DI1cvlmR+6pz3OzbOMZTdoRTZO88I/0eMgKCFBjuioSCggkDuBtR2L2FGKMxZqKda1PGIS9hbqe4b8ovq6SV/h4EoLVcGN5e1Vcnu0NurUqnzBZVDu9UZ7905WEWlCMEwrPVB1CNIjqJ33GxzXosB3cHwIQLtbQq0RErXc7IUSa1ftWPWOPkRBNtyN1rhwCwDpaMdAbtkxby6u1f6KF58bk/UhindOBzOvfemqDXeaRa0SUxC499aS4zc4nizVO7hTu2dySfnSL4RcOynyGEBufK0HxKkZnW0/TmNFJTtLgDAWWq0RfU0XQVx6UNspk93Sj1wx8xH8Ef/lentmIpQl1LiZ8j8h5Dybu0epszHDW1rYKrqGViRsQl+K3kn1B+fjR7KQBIFZKADQdm1ZekAtXrq0H9pt8XmixqkUPRiKOK79sAomcOCbmX75UzhvxvmafkeTh94WtzO4B+mbDNK8jDUbhAOHC4T62YCe11/mBA2UHNxb2zFiBrebbg6T7R8dVtW39t11RK6MLO2LC4fb/VCJNlMOoRgqVgay2UuqesXvoFdk89HxtYPOdbPjmJ7hrc1RnotTuiYd5JFz4BE0xH36K0zFfc+RWSxJ5uHW/BKVo1swNhRW9KQ4frXGhpZRm9Sxa8DlQiP3kNapZVzujoWb7NNrwh5+yCERKwFFvUjokGOTf9jnkjM0AThhOAlvSyqB1axgnp9vWiUmQ+VdoG6ayHXbog8KK++V6rwMSRI3RDtcnmQnYOzd5XYiJIYneUgd4QXor/J/M1aE8CbFPd/7FJJpxuQU6dp+0x39vLsEadV4VM/NKRI1YmSgBmIa/Wx3UQTaKTmDdRr2kNq+CSch1aCORt0sGRqtJJwTIExDH5ARqHtwbkITpBggjmXckyrKSEyG7aqbkoOoPLMLTUimbKSVVeweadOymVk96zsggWTj1z+2fAVaimjFMxIxSj65MMppP/H3+uyML3NxzFWYJFIGwleVI3oPdledY4BoLKMAKXSVmsx/UwdQvY5Q1O+iFdY7nUV5JpraTJfzntSWQ00XjTLJs9t2Y5wMo6dc/nZDC6u8Pb5L1lfTktih1FDUUqvU7jNoCCgHqk4YrPzboVNbPdBU11qZi+LDe1EkL2RWHmXmH02J+Ni6gxXXT/EvyGSEwoNkv0B9q8EiWtlN5tUmciWwxiMjSC+lZ/ImqYXJOqDnQohszBtXYSzAztZB1ToQFuU53wWNCP/hcYkPfbP6z+mXv+KYCOVSXUvzsXhiaIJk3JZWmee7RgTirrWGITV58idEa9Srf0/GuVJrb2HU5JgzbBBkZSF12ikv+bMOogw7ayOgf8O7fBpJJhe25Td4NUHZ2+bH9ZiTBEI2TNErOS1rcIY45SAvhFIpEFa2m0lnj1PJCPj+LhI+GRsVxcKXvIsl3ffFW154YThwgoOkGl7KYXJsLbJbn5NFBZt6a+Q3DJe0/BgQ/k2n9ejDHr/+3DSXT1GQbjXD+pm4KQgkwwZMPkPecoZeRuWooP5mnsbkYI/nuaMBJepdOdNa2KNCOn0FHbZKpSPtkqoNfBnN97cg99DzTlUW9xPc22yr5N2+yKdh56sx9sWMiC0KpS/rrwGBHEqV5S6t5ffZspHQNx4ja6TMtd6NooGPYBIxRK8R3W6UZ62E4MwIWVg20PGHdnEV1ys3JM48fiksQaOwM9ve6pAbQexIii3lKJYEnJOtWugYSfky5xSGiJ5uK2S4BGtfm5OgfRMVuUP6VrDny9ErQSrVzY+wbT5xuhMLA3llTE0nGIXqd57+86lWviWGBHNZrhwhfIex50IeA22b/S9ArdnqrUQ0z8y9rJdJTGAPvdTJGgYQxB1p4x4L9EJZYv1qe8FQKU8YWDwIVo+XpXkaxf/8Eigr14MQr9RFsnXgTvjuM+wBh9OKjuQsm/aoXsLlD8lqXnoZoxLF7TP8IIxAm7YaP/TJ [... cut ...] WODtCMZEYcmIlJIJrvVygacWVa92L1s6rOm0lwqZJ8yNF4DW9pJUnCsH15rEx5qTdDPRF0lnSJvCdiGZpZyYs3z+zqrccSJAH3yFrQ9UVclMRGPTJf6qg6Y+FATVcMfkvoSwoxIVrImmgwTZs1HJy6yCC5BC67h4bqkJI2x3Zm309uh6KjmLO26ghxix3gUVoOr+3yhr3kon1nHySxqAMaizAlDUZAzRdM5QX+OXbc/Jo7yHNSRdHtyqTGSJPk9gaQ/mfPZZXV0IoWwlqaJvSDbI9jSSRTigQLtBHDdl9jmrUddkeirkytuV2bSatkAdrh6W1TwrV+W7xr7bA3l2nGIVM1pn0dgIguAhh5AolUjm6ghTcvgQbWhTPGVb06KWTrROM6OERHpP87vUhlJLz2WhM+LuwZtcS5lQjQgehp3kk4Seb0Ekdi4X8A81wkautOxbL5BtVz2B2CVo4sOfgnJAVDvMnYoCfswevTPKrtU4RtZeEJ
`

mikenadeau, 2023-01-22:
I have the a similar issue with AlDente Pro 1.20 on Ventura 13.1 on a Macbook Air M1 2020.

I say similar issue because it asks for the password each time I reboot, BUT, once entered, it launches and runs fine.

* Running in the background is allowed for AlDente;
* AlDente is in the login items list.

What is it doing exactly when starting up that requires admin/write privileges?

jrittvo, 2023-02-08:
Same issue here.  Need to enter my pw after every reboot, and then start the app up from the Applications folder anyway. MB Pro M1, 13.2.  I think this started happening after upgrading to Ventura.

23727, 2023-02-11:
Same issue as well. Ventura 13.2 and Al Dente 1.20

confluencepoint, 2023-02-15:
Same issue as well. Ventura 13.2.1 (22D68) Al Dente 1.20

netique, 2023-02-15:
Same for me. I tried to reinstall the helper service with the button in the settings. That didn't help, so I removed the app completely from the Mac. That didn't help either.

MatthiasKerbl, 2023-02-18:
Hi everyone,

Thanks for reporting this bug.
We have investigated it thoroughly and will release a fix with the next update.

Cheers,
Matthias

MatthiasKerbl, 2023-03-12:
Hi everyone,

We have just released AlDente 1.21.1 which includes a fix for the issue you mentioned.
Please try it out and let me know if it works for all of you.

Cheers,
Matthias

papas68, 2023-03-12:
After 1.21.1 update asks for reinstall helper on every reboot
M1 MacBook Pro, Ventura 13.2.1
AlDente Pro Version: 1.21.1 

| George ***@***.***  ....... |

 
 

    Στις Κυριακή

MatthiasKerbl, 2023-03-12:
Hi everyone,

This issue seems to be caused by two potential reasons:

1. For some reason the helper application did not get updated and the new version of AlDente does not work with the old helper application.
2. More than one AlDente application file or installation on the system exists.

To fix these potential causes of this issue, please follow these steps:

- Open AlDente and go to Settings/General/Remove in the Helper Application section and remove the helper.
- Open the Finder and go to the Applications folder and delete the AlDente application file.
- If you have more AlDente application files anywhere else on your system, delete all of them as well.
- Go to finder and navigate to: /Library/PrivilegedHelperTools/ and delete the file: com.apphousekitchen.aldente-pro.helper if it is still there.
- Navigate to Library/LaunchDaemons/ next, and remove: com.apphousekitchen.aldente-pro.helper.plist if it is still there.
- Restart the MacBook
- Download and install the newest version of AlDente from here: https://github.com/davidwernhart/AlDente-Charge-Limiter/releases/download/1.21.1/AlDente.dmg
- Restart the MacBook to check if the bug is still happening.

Cheers,
Matthias

taylor325, 2023-03-12:
Ok. I followed uninstallation guide [here](https://apphousekitchen.com/how-to-uninstall-aldente-pro/) and redownloaded AlDente from [here.](https://github.com/davidwernhart/AlDente-Charge-Limiter/releases/tag/1.21.1)
After 2 times reboot finally AlDente working normally.
I don't know if this solves the problem permanently, I need more testing. 😎

taylor325, 2023-03-13:
Hello guys!
I have a temporaly solution: 
You need uninstall guide from [here](https://apphousekitchen.com/how-to-uninstall-aldente-pro/)
After complete uninstall you need reinstall AlDente Free or Pro.
After reinstall you need reboot and please go to mac system settings, general, login items.
You need remove AlDente from Login items and please close AlDente.
[image]
Open AlDente and you will see this notification:
[image]
Just reboot again and enjoy AlDente! 😎

MatthiasKerbl, 2023-04-12:
Hi everyone,

We have just released AlDente 1.21.4 which includes a fix for this issue.
To get AlDente 1.21.4 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Cheers,
Matthias

danieleleoni, 2023-05-03:
I already have AlDente 1.21.4 because I purchased the application few days ago, but I have this problem
Just to confirm that uninstall completely solved the issue

Zonaldo7, 2023-06-28:
I have a pro license and upgrade Aldente to the latest version and still got the problem

MatthiasKerbl, 2023-10-05:
> I have a pro license and upgrade Aldente to the latest version and still got the problem

Hi @Zonaldo7,

Do you have "Allow in the Background" enabled for AlDente in macOS settings?
[image]

Best,
Matthias

MatthiasKerbl, 2023-10-06:
> I have the same problem. Just tested now to be sure, the issue is always reproducible.
> 
> * Macbook Pro M1 Max
> * MacOS Ventura 13.6
> * AlDente Pro 1.22.2 installed via SetApp
> 
> AlDente Pro is set to launch at login and is allowed to run in background, still every time I restart my computer I get the popup to install the helper.
> 
> [image]
> I'm not sure what would happen if I marked "Don't ask again". Since it requires the password to install the helper, I guess it just wouldn't install it anymore and probably wouldn't work properly.

Hi @esimioni,

Thanks for letting us know.

Did you try the following steps?

- Please go to Finder and navigate to /Library/PrivilegedHelperTools/ 
- and delete the file: com.apphousekitchen.aldente-pro.helper  
- Then, please navigate to Library/LaunchDaemons/ and remove: com.apphousekitchen.aldente-pro.helper.plist 
- Also remove your AlDente application and reboot your MacBook.
- Then, install it as if it was freshly downloaded.

Let me know if that works.

Best,
Matthias

esimioni, 2023-10-09:
@MatthiasKerbl, it worked.
When I saw that the bug was solved, I was expecting that a new version would fix this automatically.

trss, 2024-09-21:
> Do you have "Allow in the Background" enabled for AlDente in macOS settings?

Allowing in the background solved it for me. But I don't see the connection between the two and would like to if it's not a bother.

huyz, 2024-10-06:
> Do you have "Allow in the Background" enabled for AlDente in macOS settings?

I believe this must be done for the `David Wernhart` entry as well. There are two entries total.

trss, 2024-10-07:
> > Do you have "Allow in the Background" enabled for AlDente in macOS settings?
> 
> I believe this must be done for the `David Wernhart` entry as well. There are two entries total.

I see only one entry:
[image]

MatthiasKerbl, 2026-03-11:
Hi everyone,

We have just released AlDente 1.37, which includes a fix for this issue. To update to AlDente 1.37, open the Dashboard, click on the gear icon in the bottom left corner, and click on Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2026-03-11 by MatthiasKerbl