AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #868: Can't activate Pro license on additional machine.

jcr-, 2023-01-06:
**Describe the bug**
Al Dente will not let me activate Pro  license on a third machine. Per the documentation at the time I purchased, I should be able to use it on three machines I own

**To Reproduce**
Steps to reproduce the behavior:
1. Installed Al Dente Pro on a new (third) machine
2. Wiped old machine and re-installed OS
3. Installed Al Dente pro again back on old machine
4. Attempted to activate license
5. Error dialog appears "Error: License code has already been utilized"

**Expected behavior**
Pro license works on three machines as advertised at purchase time

**Screenshots**
[image]
**MacBook Pro):**
 - MacBook model MacBook Pro Retina 15-inch mid-2015 
 - macOS version Monterey 12.6.2
 - AlDente Pro or AlDente Free? Pro
 - AlDente Version? 1.20//
 - List of all enabled/disabled settings in AlDente: Only Free ones since I can't activate Pro.

**Additional context**
Al Dente Pro was previously installed and working on this machine, but after a format of the SSD and clean system install, a re-install of Al Dente fails to activate pro license. Is it possible that this is because it was not de-activated on the machine before SSD erase? There appears to be no online way to de-register a machine to keep under the three-machine rule. Obviously I can't de-activate on the formatted drive. Also it is the same machine with the same serial so IDK.

MatthiasKerbl, 2023-01-12:
Hi @jcr-,

Usually, you need to deactivate a license in AlDente before it can be used on a newly installed AlDente.
Unfortunately, we have not yet implemented an online platform to deactivate licenses, therefore, you need to send an email to our support.

Cheers,
Matthias

MatthiasKerbl, 2023-03-12:
Hi @jcr-,

Could you resolve your issue with our support?

Cheers,
Matthias

MatthiasKerbl, 2023-03-21:
Hi @jcr-,

we haven't heard back from you in a while. I will close this issue for now.

Cheers,
Matthias

closed 2023-03-21 by MatthiasKerbl