AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #869: Aldente stops working with Ventura - can't be launched

f2107, 2023-01-09:
**Describe the bug**
After update of MacOS to ventura to 13.1 and update Aldente to 1.20 it is stop working.
I should remove old helper... when I do so... Aldente couldn't be started anymore.

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 M1 Max
 - macOS version (https://support.apple.com/en-us/HT201260)
 13.1
 - AlDente Pro or AlDente Free?
 Pro
 - AlDente Version? 
 1.20
 - List of all enabled/disabled settings in AlDente:
 ¯\_(ツ)_/¯ - can't start the app anymore

**Additional context**
no

MatthiasKerbl, 2023-01-12:
Hi @f2107,

This issue usually occurs when AlDente gets updated from a much older version.
Try to follow the uninstallation guide on our blog here: https://apphousekitchen.com/how-to-uninstall-aldente-pro/
Afterward, download and install AlDente again.

Cheers,
Matthias

f2107, 2023-01-12:
Thank you @MatthiasKerbl 
That works for me.

closed 2023-01-12 by f2107