AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #870: Temporary allow "Charging When Sleeping", after click "Top Up"

rekby, 2023-01-09:
**Is your feature request related to a problem? Please describe.**
I click "Top Up", then close my macbook for a night. At morning I found macbook with same battery level as close it. Cause: enabled mode: "Stop Charging When Sleeping".

**Describe the solution you'd like**
Temporary turn off option "Stop Charging When Sleeping" when "Top Up" enabled: after click the button I want 100% charged battery and no need restrictions in sleep mode.

**Describe alternatives you've considered**
Stay macbook with open lid with disable sleep mode, disable the option manually.

MatthiasKerbl, 2023-01-12:
Hi @rekby,

Do you have the newest version of AlDente (1.20) installed?
Because we fixed the exact issue you are facing with the newest version.

Cheers,
Matthias

rekby, 2023-01-13:
@MatthiasKerbl try with 1.20 - problem solved, thanks :)

closed 2023-01-13 by rekby