AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #872: Interface language

oleg-ezlinks, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I see AlDente interface with russian language.

**Describe the solution you'd like**
I want to be able to force English for AlDente interface.

**Describe alternatives you've considered**
Proper translations support when users decide translation to use.

MatthiasKerbl, 2023-01-12:
Hi @oleg-ezlinks,

That's a good idea. We will try to implement it with one of the future updates.

Cheers,
Matthias

MatthiasKerbl, 2023-02-26:
Hi @oleg-ezlinks,

We have looked into this and discovered that macOS already offers this functionality. Therefore, to keep everything as simple as possible, we will not add this feature in AlDente.
If you are interested in how to change the language for AlDente on macOS, we have published a tutorial on our blog here:
https://apphousekitchen.com/step-by-step-guide-change-language-for-a-single-app-on-macos/

Cheers,
Matthias

closed 2023-02-26 by MatthiasKerbl