AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #877: Control Magsafe LED feature - does not show green at 100%

dverha, 2023-01-13:
**Description**
Magsafe LED does not show green once battery tops off at 100% if the "Control Magsafe LED" option is enabled.
Also (and probably related): The AlDente battery icon shows still a + sign, indicating that charging is still in progress. Although the Apple battery icon says "Fully charged".

**To Reproduce**
Steps to reproduce the behavior:
1. enable "Control Magsafe LED"
2. top up battery or just set to 100%
3. monitor magsafe LED once battery is 100%

**Expected behavior**
Magsafe LED should show green once battery charge met the charging limit

**Screenshots**
N/A

**MacBook (please complete the following information):**
 - MacBook model Macbook Air M2
 - macOS version Ventura 13.1
 - AlDente Pro
 - AlDente Version 1.20

**THANKS**
Thank you for looking into this and thank you for this great application!!

MatthiasKerbl, 2023-01-13:
Hi @dverha,

Do you have the feature "Hardware Battery Percentage" enabled in AlDente?

Cheers,
Matthias

dverha, 2023-01-13:
Hi Matthias,

No, I have not.

Kr,

Daniel

MatthiasKerbl, 2023-01-17:
Hi @dverha,

We could replicate this issue and will try to fix it with the next update.

Cheers,
Matthias

shenzo1, 2023-03-07:
I have the same issue. Even after de-installing AlDente, the behavior remains the same: at 100%, the light does not turn green anymore. Only after unplugging and replugging the magsafe plug does the light turns from orange to green.

MatthiasKerbl, 2023-03-12:
Hi @dverha and @shenzo1,

We have just released AlDente 1.21.1 which includes a fix for the bug you mentioned.

Cheers,
Matthias

buzz8year, 2023-06-30:
No you do not!!! Why you closed it? This stupid bug still happens even after the app is removed! So you explain how to fix this now, having none of your apps installed?

buzz8year, 2023-06-30:
@dverha and @shenzo1,

have you had a chance to recover from this dumb bug?

DamonHD, 2023-06-30:
Why do you feel the need to be so rude?  What do you think it gains you?