AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #878: Prevent "Paused to save battery" messages, because Al Dente is not charging battery?

mikebore, 2023-01-15:
Certain processes eg Photos syncing, and Mailbox rebuilding are (sometimes?) automatically paused when MBA or MBP because machine thinks it is on Battery power because Al Dente is preventing charging. 

Is there an AlDente, or a global setting which will stop this happening? 

Thanks for great app!

MatthiasKerbl, 2023-02-01:
Hi @mikebore,

Usually, when AlDente paused charging, the MacBook does not believe it is on battery power. It is just not charging further. As seen in the following screenshot:
[image]

Are you still experiencing some background processes being paused due to the "Paused to save battery" message?

Also, during the Discharge mode, the MacBook actually switches battery power.

Cheers,
Matthias

mikebore, 2023-02-01:
I noticed the problem while setting up a new computer which involved a lot of Photos and email syncing. The issue has not recurred since I finished those.

Anyway your reply confirms there isn't a setting I have missed.

closed 2023-02-02 by MatthiasKerbl