AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #879: Visual feedback on no power

sandr8, 2023-01-16:
In certain countries, power plugs are really easy to fall off the socket and disconnect, hindering all of one's effort to minimize the battery cycles.

For macbooks with no magsafe connector, it would be extremely nice to have some hard to miss visual clue that this has happened.

Something much more obvious than the tiny menubar icon would be for the entire menubar or dock background to switch colors.

Thank you!

MatthiasKerbl, 2023-01-23:
Hi @sandr8,

I understand where you are coming from and I had the same issue before as well. However, I am having a hard time seeing how this could be implemented without being annoying. It could be a notification that the MacBook got unplugged, while it was charging. This also means that we would get a notification every time we are doing it on purpose.
How would you like this to be implemented?

sandr8, 2023-02-03:
> the entire menubar or dock background to switch colors

I was thinking of something a la [menu bar tint](https://manytricks.com/menubartint/). One could have one color menubar when plugged in, and have it automatically switch to another color when unplugged. I for example would choose black for plugged and a dark merlot for unplugged. I don't think I would find it annoying.