AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #882: Calibrating leaves the limit at 34%

madler, 2023-01-18:
I ran a calibration, and when done (none of the steps bolded), the charge limit was left at 34%, which is the final discharge level. Is this the expected behavior?

I would have expected it to return the charge limit to what it was before the calibration. (I had it at 80%.)

[image]

[image]

MacBook Pro (14-inch 2021)
macOS 12.6.2
AlDente Pro 1.20
Sailing, battery icon, % icon, control LED, blink orange all enabled

MatthiasKerbl, 2023-01-18:
Hi @madler,

This is not the expected behavior.
When the Calibration Mode finishes, it will discharge to the set charge limit from before the calibration. Furthermore, during the Calibration Mode, the active state is bolded.
Did you try it again and are you still experiencing this behavior?

Cheers,
Matthias

madler, 2023-01-18:
The active states were in fact bolded when active. I provided the screen shot to show that none were bolded, indicating that the calibration was done.

I think you mean it will _charge_ to the set charge limit from before the calibration. I doubt that anyone has their charge limit set to below 34%.

Ok, I'll run it again. But it takes forever.

madler, 2023-01-18:
Now as I am re-running the calibration I see this:

[image]

It is now indeed planning to _discharge_ to 80% once it's done, which is what I had it set to before the calibration.

I think I know what happened. At some point during the discharge of the first calibration, I accidentally knocked the magsafe connector off. I immediately put it back on. The interruption of power stopped the calibration (even though it was discharging at the time and presumably not using the power). I restarted the calibration. I am guessing that at that time, the charge was 34%, which, for whatever reason, it took as the target for the end of the restarted calibration. I hadn't noticed either end target at that time, and when the last step was a discharge fo 34%, I assumed that that was part of the process, even though it was an arbitrary-looking number, and that it would go back to 80% after that. When in fact it thought that 34% was the charge limit I had set. Which it wasn't.

So the bug is that when a calibration is interrupted or restarted (not sure which), it forgets what I had set the charge limit to, and instead picks up the current charge level. It should use the last charge limit that I, the user, set.

madler, 2023-01-20:
It did indeed finish up at 80%. Not 34% like the time before.

MatthiasKerbl, 2023-01-20:
Hi @madler,

I tried to replicate the potential bug you encountered but was unsuccessful.
When I unplug the MacBook during the first discharge during the Calibration Mode, I get the following message from AlDente:

[image]

When I click on "OK", the Calibration Mode continues as intended and the set Charge Limit does not change.
If I click "Cancel", the Calibration Mode gets disabled and AlDente charges to the set Charge Limit. When I activate the Calibration Mode again, it runs as expected and uses the user-set Charge Limit.

Do you know how to reliably replicate the potential bug you encountered?

Cheers,
Matthias

MatthiasKerbl, 2023-03-12:
Hi @madler,

Any updates on your issue?

Cheers,
Matthias

MatthiasKerbl, 2023-03-15:
Hi @madler,

I am assuming the Calibration Mode works now as expected.
Since you have not reported back in a long time, I will close this issue for now. However, feel free to get back and I will open the issue again.

Cheers, 
Matthias