AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #885: Prompted to install update, Macbook Pro now entering lock screen every five seconds.

tchpanda, 2023-01-23:
**Describe the bug**
I was prompted to install the latest update today (23 Jan).  Upon agreeing to do so my Macbook Pro enters lock screen after a only few seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. Accept the latest update.
2. Observe maddening screen locking behaviour.

**Expected behavior**
Nothing.

**Screenshots**
N/A - I have removed this software.

**MacBook (please complete the following information):**
 - MacBook Pro 16-inch, 2021
 - macOS Ventura 13.1
 - AlDente Pro
 - AlDente Version - whatever has been released in the last day.
 - List of all enabled/disabled settings in AlDente:  N/A I have had to uninstall.

**Additional context**
Add any other context about the problem here.

tchpanda, 2023-02-04:
Hi Matthias,

Sorry for the delayed response.  Reinstalled the helper and all appears fine now.

MatthiasKerbl, 2023-02-04:
> Hi Matthias,
> 
> Sorry for the delayed response. Reinstalled the helper and all appears fine now.

Hi @tchpanda,

I am glad everything works as expected again.
I will close this issue for now but feel free to report back if anything is off again.

Cheers,
Matthias

closed 2023-02-04 by MatthiasKerbl