AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #887: Charge permanently limited won't allow charging even when battery as low as 2%

hemicyon, 2023-01-28:
**Describe the bug**
Sporadically, AlDente seems to permanently limit charging and won't allow the battery to charge at all to the point the battery drains completely. This has happened to me in standard and Intel modes. Occasionally, AlDente will also fail to limit the charge to 80% and will charge all the way to 100% regardless, but I'm not sure if these two things are related. Somehow it even convinced my Mac that it wasn't plugged in at all; AlDente showed the power was connected, but Mac gave me a warning to plug in and none of the other battery reading utilities I have showed that the Mac was plugged in.

Today, I was working on my Windows 10 partition (booted through boot camp) when suddenly I was given a battery warning and urged to plug my MacBook in, but of course it was already plugged in—it's always plugged in. Saved and quit everything, and the battery did indeed show 10%, going down rapidly to 7% while I was looking at it and preparing to reboot into macOS.

Logged into macOS and the battery still showed near empty, charging paused. While investigating, AlDente kept switching back and forth between charging and limiting (though it was usually limiting). I tried fiddling with the charge limit, and it would briefly show charging with each change and then immediately go back to limiting. I tried clicking the top up button, and again it would show charging briefly only to immediately go back to limiting. Charge got down to 2% during these quick tests. 

Shut down the Mac, unplugged everything, left it for a few minutes, then plugged it back into power only. The Mac booted up automatically (so the Mac recognised it had been plugged in), but AlDente was still limiting the charge once macOS loaded (even before I logged in). Set the limit to 100%, disabled launch on login, quit AlDente, shut down again, did the SMC reset, booted back up and the battery is charging normally again. Although I'm a little too afraid to launch AlDente right now to confirm all the settings (though I'm fairly sure I didn't have anything enabled at the time). Intel mode was not enabled today, but all of this has happened before when Intel mode was enabled. 

I stopped bothering with Intel mode when it seemed like standard mode was still able to limit the charge to roughly 80% when I was booted into Windows, although this stopped working recently, and Windows will always charge my battery to 100%. But it got to be too much of a hassle to remember to turn Intel mode on and off between partition boots, so I decided it was fine for the battery to charge to 100% while I'm in Windows.

**To Reproduce**
Steps to reproduce the behavior:
I'm not sure because it happens out of nowhere. I work for several hours in macOS, then will reboot into Windows and work there for several hours. After that I may either reboot back into macOS and work a bit more there (after which the computer will sleep) or just shut the computer down entirely.

**Expected behavior**
I expect AlDente to work as advertised, limit the charge to 80% (or whatever it is set to) and allow the battery to charge when it dips below 80%, preferably in both macOS and Windows partitions, but in macOS at the very least without any adverse effects in Windows.

**Screenshots**
I grabbed this screenshot while fiddling with the limit—I don't know what that little fire symbol is or why it's there in the bar because it usually isn't
[image]
This just shows my other battery utilities and what they show during this, plus Mac's notification to plug in even though the Mac was definitely already plugged in
[image]

**MacBook (please complete the following information):**
 - 2019 MacBook Pro 16" (MacBookPro16,1)
 - macOS 12.6.1
 - AlDente Pro (via SetApp)
 - AlDente v1.20
 - List of all enabled/disabled settings in AlDente: can't see all the settings right now because I won't launch the app right now while the battery is coming up, but I'm pretty sure I had everything disabled except for *maybe* automatically discharge? I can't remember if I had disabled that recently or not

**Additional context**
I completely uninstalled and reinstalled AlDente two times before when this happened the last time. I'm not sure what's going on, but it's terribly jarring and disruptive.

DamonHD, 2023-01-28:
Is the flame icon indicating that your system is too hot and thus the battery is not being charged (Heat Protection)?  Maybe you have accidentally set a very low value for that?  I have 35C for mine.

Rgds

Damon

hemicyon, 2023-01-28:
Interesting, when I launched AlDente just now, both sailing mode and heat protection (35°C) were enabled (and nothing else). Except I would have sworn that I disabled both of those settings. Heat protection is generally preferable, but I'd rather the battery be hot and charged/ing rather than hot and discharging completely in the middle of working. Not sure how they came to be enabled on their own. But however it happened, that would seem to explain the weird behaviour in macOS with the battery not charging, but I'm not sure that explains how the battery came to drain completely in Windows since it usually charges clear to 100% there lately.

DamonHD, 2023-01-28:
I think that a recent upgrade may have changed user settings: for example my sailing % went from 5 to 1 all by itself.

Rgds

Damon

larissa-n, 2023-02-19:
Hey @MatthiasKerbl - I've always had Heat Protection disabled and still, it's happened a few times now that Al Dente will keep my Mac from charging and, if I don't pay attention, drain it until it shuts itself down. Just had to reset my SMC to even get it to start again. This definitely seems like a bug.

closed 2024-01-24 by MatthiasKerbl