AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #889: Location / charger based configuration

adriangb, 2023-01-30:
Generally I want to keep my charge at 70% when I'm at home plugged into a dock, which is 90% of the time. The other 10% of the time I'm traveling and would actually prefer to charge up to 100%. It would be really nice if AlDente supported some basic rules for configuration. It'd be even better if it could also control other configurations like sleep settings or Apple's adaptive battery charging.

> When I am plugged into <some charger identifier> and am at <work or home> I want to limit charging to 70%, turn off Apple's built-in charge management and never sleep when the lid is closed. Any other time I would like to allow charging up to 100%, turn on Apple's charge management and sleep when the lid is closed.

I realize a lot of this goes beyond the scope of AlDente but I wanted to give you the full idea and let you pick pieces that make sense (if any).

MatthiasKerbl, 2023-03-12:
Hi @adriangb,

Interesting ideas but for now it seems a little bit too niche of a use case for us to focus on implementing it in AlDente. However, since AlDente Pro supports Apples Shortcuts I am pretty sure you can set up location-based rules for AlDente through the Shortcuts app already now.

Cheers,
Matthias

adriangb, 2023-03-12:
Seems to work great, that's good enough for me. I had never used Apple Shortcuts / missed that in the docs. Thanks!

closed 2023-03-12 by adriangb