AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #893: Issue loading external hard drives while AlDente is open and running

beamer1433, 2023-02-03:
I tried connecting external hard drives to my MacBook Pro 2021 M1 Max chip , via usb-c and they weren't mounting and causing a beach ball in Finder and in disk utility. The hard drives were encrypted and it wouldn't let me enter the password . I tried different USB A to USB -C adapters and different hard drive docks, but all did the same thing. Then I tried randomly quiting AlDente , and all of sudden I was able to unlock the disk and mounted instantly. I am running Mac OS Ventura 13.2

MatthiasKerbl, 2023-11-15:
Hi @beamer1433,

Do you still experience this issue?
You are the first and only user so far to report an issue with an external hard drive and AlDente.

Best,
Matthias

lgstern, 2024-02-12:
I must be the second. I have finally figured out why I have been having so many hard times with external drives.

MatthiasKerbl, 2024-02-20:
Hi @lgstern and @beamer1433,

Unfortunately, I am not able to reproduce your issue. External hard drives work flawlessly for me even with AlDente running.

Please answer the following question so that we can assess this issue better:

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

Best,
Matthias

closed 2024-08-12 by MatthiasKerbl