AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #895: Calibration didn't stopped as should

Volodymyr-13, 2023-02-05:
I started calibration and it didn't stopped at 81%, is counting it's discharge.

[image]

[image]

I stopped it on that moment, maybe it will check % from the system aka not real hardware battery percentage?

MatthiasKerbl, 2023-02-10:
Hi @Volodymyr-13,

Thanks for the bug report. You are correct, in AlDente 1.20 we are using the macOS battery percentage for the Calibration Mode even if the Hardware Battery percentage is activated. However, we already fixed it and will release it with the next update.

Cheers,
Matthias

MatthiasKerbl, 2023-03-13:
Hi @Volodymyr-13,

We have just released a new version of AlDente Pro which includes a fix for this issue.

Cheers,
Matthias

closed 2023-03-13 by MatthiasKerbl