AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #898: Turn on the starting up  auto-start option and it will not take effect

dongxr, 2023-02-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2023-02-18:
Hi @dongxr,

Just to be sure, you have Launch at Login enabled but AlDente does not launch automatically when you restart your MacBook?

dongxr, 2023-02-26:
> 嗨@dongxr，
> 
> 可以肯定的是，您启用了登录时启动，但重新启动MacBook时AlDente不会自动启动？

yes mbp14 macos12.6.3 aidente1.20

dongxr, 2023-02-26:
> Hi @dongxr,
> 
> Just to be sure, you have Launch at Login enabled but AlDente does not launch automatically when you restart your MacBook?

 hi @MatthiasKerbl 
yes 
mbp14 2021 
macos12.6.3 
aidente1.20

hadleyac, 2023-03-08:
I'm also experiencing this issue. I tried manually removing and re-adding it to login items with no luck. I have also completely uninstalled the app/helper and reinstalled.

MatthiasKerbl, 2023-03-12:
Hi @dongxr and @hadleyac,

We have just released AlDente 1.21.1 which has many stability improvements.
Let me know if you are still experiencing this issue with the newest version.

ComicDansMS, 2023-04-02:
I have an issue where AlDente is sometimes dropping out of the login items. It may or may not be related to this issue here, but I wasn't sure if it was worth opening a new one.

It could be due to updates being installed, but I can't be 100%. Sometimes I will come to my mac, find it has restarted itself (I usually just let it sleep), and the battery is at 100% without Aldente running. I then find that AlDente is not in the login items. This has happened a few times now.

Is there anything that could be causing this? I'm thinking it might be a macOS issue rather than Aldente related.

MatthiasKerbl, 2023-04-08:
> I have an issue where AlDente is sometimes dropping out of the login items. It may or may not be related to this issue here, but I wasn't sure if it was worth opening a new one.
>
> It could be due to [...]

Hi @ComicDansMS,

Thank you for reporting your experience.
We have investigated this issue and are confident that we were able to fix it. It will be released with the next update.

Cheers,
Matthias

MatthiasKerbl, 2023-04-25:
> Hi Matt,
> 
> "Launch at login" is enabled, but after I restarted the OS, AIDente was not auto restarted.
> 
> I have tried many times and it never works.
> 
> I checked "System Settings", AIDente has been added at "Login Items" -> "Open at Login".
> 
> BTW, the OS user I am using to login is a regular user, it doesn't have admin permission, not sure if this is related to issue.
> 
> Thanks, Richard

I am sorry that you are still experiencing this issue.

Could you try the following steps and report back if it helped?

1. Disable "Launch at login" in the AlDente Settings.
2. Restart the MacBook.
3. Enable "Launch at login" again.
4. Check if AlDente launched.

Cheers,
Matthias

Zonaldo7, 2023-06-30:
still got this problem
mbp m1 pro
aldente pro 1.22

Nshandeshu, 2023-07-07:
still got this problem

Shankar1598, 2024-02-05:
I am facing this problem as well

Zonaldo7, 2024-02-05:
Simply reinstalling helps solve this problem well.

MatthiasKerbl, 2025-02-21:
Hi everyone,

For some users, the following steps helped:

1. Make sure the AlDente app file is in the Application folder of macOS.
2. Deactivate and activate the launch at login setting within the AlDente settings.
3. Restart the MacBook.

Best,
Matthias