AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #899: MacBook Pro 14" M1 won't turn on if the battery runs out, using AlDente

alecrobertson1, 2023-02-19:
**Describe the bug**

My bug is very similar to this one: https://github.com/davidwernhart/AlDente-Charge-Limiter/issues/491.

I have set the charge limit to 75%, if the MacBook Pro runs out of battery to the point where it turns off to save power, the laptop will not turn back on again when given power. I have to leave the MacBook Pro plugged in for over an hour, at which point it will become responsive again. This only occurs when using AlDente.

When this is happening, the MagSafe cable will blink orange: on, off, on, off.

**To Reproduce**
Steps to reproduce the behavior:

Set charge limit to 75%
Allow MacBook Pro to run down and then turn off
Leave for some minutes and observe when given power, the MacBook Pro is unresponsive.

**Expected behavior**
A clear and concise description of what you expected to happen.

The MacBook Pro should not become unresponsive when it runs out of battery.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro (14-inch, 2021)
 - macOS version: 12.6.3
 - AlDente Pro or AlDente Free? AlDente Pro
 - AlDente Version? AlDente 1.20
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2023-03-15:
Hi @alecrobertson1,

Thank you for the detailed report and I am sorry you are experiencing these issues.

It sounds like you have a severe case of battery miscalibration.
First, to avoid this in the future, fully charge and discharge your battery at least every two weeks. Furthermore, in such a severe case of miscalibration, AlDentes Calibration Mode does not help currently. It prevents the battery from becoming miscalibrated.
To get your battery out of this state, try the following:

1. Charge your battery to 100% and leave it plugged in for 1h+ after it reached 100%.
2. Unplug your MacBook and let it discharge as slowly as possible, the best is to just leave it idle and avoid letting it go to sleep.
3. When the MacBook shuts down again or the battery is empty, repeat steps 1 and 2 as often as necessary to calibrate the battery again and it normally discharges to <5% again. Sometimes, this can take 5+ cycles.

Hope that helps.

Cheers,
Matthias

MatthiasKerbl, 2023-04-15:
Hi @alecrobertson1,

Any updates about your issue?

Cheers,
Matthias

alecrobertson1, 2023-04-20:
Hi @MatthiasKerbl, I've followed the steps above and so far it has not re-occured.

Many thanks for your swift resolution.

closed 2023-04-21 by MatthiasKerbl

Aghyads, 2023-04-21:
@MatthiasKerbl 

Dear Matthias, 
I would like to purchase your app ‘AlDente’. Therefore, I was wondering if you can offer me any discount code that I can use 😊 or maybe the opportunity to have two life time license instead of one, so I can install the app on my wife’s macbook as well. 
I hope this can be possible! That would be so much appreciated!

Thank you so much in advance, and I wish you a wonderful day.

Kindest regards, 
Aghyad

MatthiasKerbl, 2023-04-21:
Hi @Aghyads,

Thank you for your interest in AlDente Pro. :)
Unfortunately, we do not offer any kind of discount at the moment. However, if you buy AlDente Pro, you can activate it on up to 3 of your devices.

Cheers,
Matthias

MatthiasKerbl, 2023-04-21:
> @MatthiasKerbl, Fair enough :) Then I will purchase the Pro! Any instructions' page where you guys have the best settings that keeps the battery at the best life cycle can be? Cuz there are many info on the internet, but what you advice your client? Thnxx in advance for your time!

Hi @Aghyads,

You should start with our [blog](https://apphousekitchen.com/blog/).
In general, we are trying to be careful with giving too many suggestions because usually it always depends on the individual situation and it is not always clear which specific setting is the best one. Therefore, we always recommend doing your own research as well.

Cheers,
Matthias