AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #906: Charge limit failure

hhrensheng, 2023-03-10:
Hello I am a foreigner。

[image]

macOS:13.2.1
AlDente pro 1.20

**The following method doesn't work for me**
[image]

MatthiasKerbl, 2023-03-12:
Hi @hhrensheng,

Do you have Apple's Optimized Battery Charging disabled?
Did you try to reinstall the helper application in AlDente Settings/General/Reinstall?
Did you try to reinstall AlDente Completely? https://apphousekitchen.com/how-to-uninstall-aldente-pro/

Hope that helps.

Cheers,
Matthias

hhrensheng, 2023-03-14:
Thank you.
After I updated the software to the latest version, it worked fine.
But I found another phenomenon：
During battery calibration，A prompt will be triggered: the power is disconnected, whether to continue battery calibration？
I don't think this prompt should appear。

MatthiasKerbl, 2023-03-14:
> Thank you. After I updated the software to the latest version, it worked fine. But I found another phenomenon： During battery calibration，A prompt will be triggered: the power is disconnected, whether to continue battery calibration？ I don't think this prompt should appear。

Hi @hhrensheng,

I am glad it works now. :)
You are right, we should adapt the notification for this scenario.

Cheers,
Matthias

closed 2023-03-14 by MatthiasKerbl