AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #908: MagSafe3 LED control fails after upgrading from 1.20->1.21

dbavatar, 2023-03-11:
**Describe the bug**
After upgrading to 1.21 MagSafe only blinks green at approximately 2 second intervals.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade from 1.20 to 1.21
2. Leave "Control Magsafe" enabled alone.

**Expected behavior**
Solid amber during charging, solid green when not charging

**MacBook (please complete the following information):**
2021 Macbook Pro M1 Max MacBookPro18,2
Ventura 13.2.1
AlDente Pro
1.21

PPFilip, 2023-03-11:
Same here - M1 Max running on  13.2.1 (22D68) , AlDente Pro 1.21

lakoliu, 2023-03-11:
Same here, just updated and this started happening right away. It worked as expected in 1.20.

furrcodes, 2023-03-11:
Same here - 14" M1 Pro, MacOS 13.2.1 (22D68), AlDente Pro 1.21

egehandr, 2023-03-11:
Same here - 14" M1 Pro, MacOS 13.2.1 (22D68), AlDente Pro 1.21

MatthiasKerbl, 2023-03-11:
Hi everyone,

Thanks for sharing this bug and sorry for the inconvenience. We will try to release a fix as soon as possible. Until then it might be best to keep "Control MagSafe LED" disabled.

Cheers,
Matthias

MatthiasKerbl, 2023-03-12:
Hi everyone,

We have just released AlDente 1.21.1 which has a fix for the mentioned bug.

Cheers,
Matthias

PPFilip, 2023-03-12:
Thank you! I can confirm that it is fixed on my side.

MatthiasKerbl, 2023-03-12:
> Thank you! I can confirm that it is fixed on my side.

Perfect.

closed 2023-03-12 by MatthiasKerbl