AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #910: After 1.21.1 update asks for reinstall helper on every reboot

taylor325, 2023-03-12:
After 1.21.1 update asks for reinstall helper on every reboot
M1 MacBook Pro, Ventura 13.1
AlDente Version: 1.21.1 (Free)
[image]

papas68, 2023-03-12:
and the same thing happens to me

AlDente Pro Version: 1.21.1

taylor325, 2023-03-12:
Ok. I followed uninstallation guide [here](https://apphousekitchen.com/how-to-uninstall-aldente-pro/) and redownloaded AlDente from [here.](https://github.com/davidwernhart/AlDente-Charge-Limiter/releases/tag/1.21.1)
After 2 times reboot finally AlDente working normally.
I don't know if this solves the problem permanently, I need more testing.

MatthiasKerbl, 2023-03-12:
Hi everyone,

This issue seems to be caused by two potential reasons:

1. For some reason the helper application did not get updated and the new version of AlDente does not work with the old helper application.
2. More than one AlDente application file or installation on the system exists.

To fix these potential causes of this issue, please follow these steps:

- Open AlDente and go to Settings/General/Remove in the Helper Application section and remove the helper.
- Open the Finder and go to the Applications folder and delete the AlDente application file.
- If you have more AlDente application files anywhere else on your system, delete all of them as well.
- Go to finder and navigate to: /Library/PrivilegedHelperTools/ and delete the file: com.apphousekitchen.aldente-pro.helper if it is still there.
- Navigate to Library/LaunchDaemons/ next, and remove: com.apphousekitchen.aldente-pro.helper.plist if it is still there.
- Restart the MacBook
- Download and install the newest version of AlDente from here: https://github.com/davidwernhart/AlDente-Charge-Limiter/releases/download/1.21.1/AlDente.dmg
- Restart the MacBook to check if the bug is still happening.

Cheers,
Matthias

MatthiasKerbl, 2023-03-12:
Also, we already have a Github issue open for this bug here #866.
Therefore, I will close this issue and post updates on the other one.
Please also share now findings on the other issue.

Thanks,
Matthias

closed 2023-03-12 by MatthiasKerbl

taylor325, 2023-03-12:
After my computer power off and reboot still got this bug.
[image]

taylor325, 2023-03-12:
> Ok. I followed uninstallation guide [here](https://apphousekitchen.com/how-to-uninstall-aldente-pro/) and redownloaded AlDente from [...]

After my computer power off and rebooted still got this bug.
I need install helper on every reboot. Weird! Never got this bug before AlDente 1.20.
[image]

papas68, 2023-03-13:
I also get the same error

M1 MacBook Pro, Ventura 13.2.1
AlDente Pro Version: 1.21.1

papas68, 2023-03-13:
the same to me

| George ***@***.***  ....... |

 
 

    Στις Δευτέρα

MatthiasKerbl, 2023-03-13:
Hi @papas68,

Do I understand correctly that you could fix the issue as well?

papas68, 2023-03-13:
I have followed the instructions but the result is the same, it also reports the same taylor325

| George ***@***.***  ....... |

 
 

    Στις Δευτέρα

MatthiasKerbl, 2023-03-13:
That's unexpected, thank you for sharing and we will investigate further.

taylor325, 2023-03-13:
After several reboot I have this bug again :( 
[image]
Asks for reinstall helper on every reboot 🤨

papas68, 2023-03-14:
the same boring for me

aldente pro 1.21.3MacBook Pro (13-inch, M1, 2020)

| George ***@***.***  ....... |

 
 

    Στις Τρίτη