AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #917: Menu item is always centered, "under" the notch

tisba, 2023-03-15:
**Describe the bug**
Menu item sits in the middle of the menu bar, under the MacBook Pro's notch.

**To Reproduce**
Just start AlDente. The menu items appears in the same position every time.

**Expected behavior**
I expect the menu icon to be "sorted in", like it normally would.

**Screenshots**

(note that the notch is not shown on screenshots)

[image]

**MacBook (please complete the following information):**
 - MacBook model: MacBookPro18,3
 - macOS version: 13.2.1
 - AlDente Pro or AlDente Free? Free
 - AlDente Version? MacBookPro18,3
 - List of all enabled/disabled settings in AlDente: Default settings, nothing changed.

MatthiasKerbl, 2023-03-15:
Hi @tisba,

Thank you for this bug report.
Which version of AlDente are you using?

tisba, 2023-03-15:
@MatthiasKerbl arg, damn! For some reason CMD+c does not work in the About dialog of AlDente 🤦 I'm on 1.21.3 (38)

tisba, 2023-03-15:
One more observation: I just noticed, that on a 2nd, external screen, the menu icon is correctly placed.

MatthiasKerbl, 2023-03-16:
Hi @tisba,

We are investigating now and will let you know if we need more details or have any updates.

MatthiasKerbl, 2023-03-20:
Hi @tisba,

Usually, you can change the location of the menubar items per drag and drop when you have the command button pressed. Are you able to just move the AlDente icon over to the right side while you have the command button pressed?
Another interesting thing for us to know would be what happens when you change the display scaling setting. Please open the macOS settings, go to Display, and switch to "More Space". What happened with the AlDente icon? Did it stay in the middle or did it move somewhere else?

Cheers,
Matthias

tisba, 2023-03-22:
Damn, that's something I could have thought of myself 🤦 Here is a recording of me dragging the item on the build-in display with the notch. It was a bit tricky, as the icon was not exactly centred - that's why I opened the app window in the beginning to get an indication:

https://user-images.githubusercontent.com/75156/226838189-5311599a-cd6c-4672-b36b-ebc5eb475f69.mov

The icon position seems to be persisted now and sticks to the right.

Changing the display scaling does not change anything now, but I'm not sure if that would have an impact on the icon position. Unfortunately I tried the dragging first…

MatthiasKerbl, 2023-03-22:
Hi @tisba,

I am glad to hear that it worked for you and don't worry about the scaling setting.
I will close this issue for now but let me know if the icon does not stay where it should.

Cheers,
Matthias

closed 2023-03-22 by MatthiasKerbl