AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #919: Top Up charge greyed out

gijap, 2023-03-16:
The Discharge button works. Top up button next to it is greyed out.

Why is this? 

**To Reproduce**
Steps to reproduce the behavior:
1. click on Aldente icon

**Expected behavior**
Top Up feature

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608) Macbook pro 14-inch, 2021,   Model Identifier:	MacBookPro18,3
  Model Number:	G15J0B/A
 - macOS version (https://support.apple.com/en-us/HT201260) 13.1 
 - AlDente Pro or AlDente Free? Free
 - AlDente Version? 1.213

MatthiasKerbl, 2023-03-16:
Hi @gijap,

The Top Up feature is only available in AlDente Pro. You can read more about the differences between AlDente Free and Pro here: https://apphousekitchen.com/

Cheers,
Matthias

closed 2023-03-16 by MatthiasKerbl

ADTC, 2023-06-28:
@MatthiasKerbl I suggest showing it somehow, like a pop-over or tooltip that says "Get Pro to enable this feature"

Same can be done for disabled toggles in the settings.