AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #921: Quickly toggle between normal and Intel mode + disable Intel mode on launch?

martin-braun, 2023-03-21:
**Is your feature request related to a problem? Please describe.**
I like the sailing feature, but I also have to boot into BOOTCAMP sometimes. I wish to quickly switch to Intel mode before booting into BOOTCAMP. 

**Describe the solution you'd like**
Please, allow me to enable a quick toggle here to enable / disable Intel mode: 
[image]

Additionally, I wish I could tell AlDente to automatically disable Intel mode when I boot back into macOS, so when launching AlDente.

**Describe alternatives you've considered**
Keeping Intel mode all the time, but I'd miss sailing. It's annoying to remember to turn it off when being back in MacOS. Also it's annoying to always go into the preferences to change the setting.

MatthiasKerbl, 2023-03-22:
Hi @martin-braun,

I understand that these changes make perfect sense in your case but I feel like it is too niche of a use case to make these changes in general. We might look into adding the Intel Mode to Apple Shortcuts. Then, you could configure it the way it fits your use case.

Cheers,
Matthias

martin-braun, 2023-07-16:
@MatthiasKerbl Okay, how about this one then: Allow me to specify a stop and start charge % for intel mode. My primary problem about Intel mode is that I disconnect the cable and put the Mac into sleep, when I come back it lost only 1%. On Intel mode it will recharge just 1% when I reconnect, which stresses the battery for no reason.

I don't want to disable charging while sleeping and even then, battery is lost a little while sleeping. I don't need sailing, but I want it to say to stop at 83%, but only start charging at 77%.

Please consider a solution. I was very perplexed by your rejection, because it would be so easy to implement, although I respect that decision, because this product is not paid on a subscription model. I paid for the software that I tested prior, so I knew what bang I'd get for the buck.

Kudos anyways.

closed 2024-06-18 by MatthiasKerbl

martin-braun, 2024-06-18:
Excuse me @MatthiasKerbl, how is this an improvement? You effectively removed a feature and force me into Intel mode. Would you please re-enable the smarter mode on Intel-Macs?

Btw, I'm not even using BOOTCAMP anymore. The new update is a step backwards in my honest opinion. Originally, I bought the app for the sailing feature which is now not available anymore.

MatthiasKerbl, 2024-06-19:
> Excuse me @MatthiasKerbl, how is this an improvement? You effectively removed a feature and force me into Intel mode. Would you please re-enable the smarter mode on Intel-Macs?
>
> Btw, I'm not even [...]

Hi @martin-braun,

I think there might be a misunderstanding. With the new update, we were able to combine the strengths of both modes. This means that all "smarter" features such as Sailing Mode, Heat Protection, Calibration Mode, and so on are working as before with the added benefit that the charge limit now has the convenience and consistency of the Intel Mode which means it will stay active during sleep, powered off, and Bootcamp.

Hope that explains the update better.

Best,
Matthias

martin-braun, 2024-06-19:
Hi @MatthiasKerbl, thanks for explanation, now the goal of the new version do actually make sense. You essential try to enforce the new mode with all the great capabilities, but switch to intel mode during sleep, power off and Bootcamp automatically, giving us the best of both worlds.

However, I have to report that Intel mode is currently enforced for me all the time and I cannot change the mode anymore. I'm on the go right now, so not plugged in, so it's hard to share screenshots, but essentially:

- In the popover I see the Intel symbol
- Under Charge and Feature options, everything is disabled, as it was on Intel mode in the previous app as well

I already restarted the device numerious times, but it would not switch to the new mode. Can I somehow assist in finding the issue?

Here is my debug file: https://psitransfer.clampcloud.com/6e24d572fb7f
The password is `AlDente-Charge-Limiter`. Thanks.

MatthiasKerbl, 2024-06-20:
Hi @martin-braun,

Thanks for your feedback and I am sorry to hear that your installation somehow glitched through the update process because in general, it should not be an issue at all to have the Intel Mode enabled before updating.
Anyways, you should easily be able to fix it by performing a setting reset in AlDente Settings/General/Reset Settings.

Best,
Matthias

frewper, 2024-09-01:
I have been Using Bootcamp with Windows 11 and am plagued with the battery issue. I have been trying AlDente to limit the battery charge, but alas it only works on the MacOS and when I restart into windows it doesnt work and charges the battery to 100%. (Limit set at 70%).  I am using Version 1.27.3 which means that Intel Mode is removed. Only thing I am using AlDente Free, so I am not sure if this is a version issue. 

If it not a version issue, what am I missing ?