AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #922: Battery level sometimes exceeds charge limit by a large margin (M1 MBA)

xezrunner, 2023-03-23:
**Describe the bug**
While Al Dente is running and the menu bar's Battery dropdown says `Power Source: Power Adapter -- Battery Is Not Charging`, over the course of some hours, the battery level occasionally jumps above the limit.

In my case, I'm limiting the charge to 70%. Using the machine for an hour or two, it trickles up to anywhere from 73 to 76%, even though macOS says the battery is not charging.
Battery calibration is being done as usual.

**Expected behavior**
Not sure if this is intended behavior (macOS battery reporting being inaccurate?)

**MacBook:**
 - MacBook Air (M1 2020)
 - macOS Ventura 13.2.1
 - AlDente Free?
 - AlDente 1.21.3?

**Additional context**
I've been seeing this happen multiple times throughout the week. So far, the worst it has been is it going up to 76% with a 70% limit. The machine was on and being used the entire time.

winkler1, 2023-03-25:
This started happening to me after a recent upgrade. M1 Mac. Really annoying.
MacBook Pro 2021, M1 Max
macOS Ventura 13.2.1
AlDente Free
AlDente 1.21.3

... are there older versions available? It broke recently.

Brog33, 2023-04-23:
Same problem here. Been going on for months. Most of the time it stops at 80% which it is set to, but a few times it stops at ~90%.

MB Pro 2021 M1 Pro 10 core
AlDente 1.21.4 Free

winkler1, 2023-04-23:
Yeah... I've stopped taking upgrades until this is confirmed fixed. Uninstalling to downgrade was a pain.

Brog33, 2023-04-26:
It works to press the discharge button when it charges over the set %, but it's a bit annoying yeah.

MatthiasKerbl, 2023-08-07:
Hi everyone,

I have tried to reproduce this issue with our newest version AlDente 1.22.2 free. Unfortunately, even after days, I am not able to experience an increase in the state of charge.
Are you still experiencing this issue with our newest version?

Best,
Matthias

Brog33, 2023-08-08:
it's still happening from time to time, but not as much as before. i will try to see if i can find a pattern when it happens.
if i have understood this, the aldente service needs to "survey" system app at all times to keep it from charging. while on my linux laptop i can set the limiter in the kernel, which makes the limiter work even when the machine is turned off.

Libroru, 2023-10-01:
Unplugged my macbook two days ago to the same 85% of battery charge even though I set my limit to 80%. Would upgrading to the pro version and using the discharge feature fix that issue?

MatthiasKerbl, 2023-10-01:
> Unplugged my macbook two days ago to the same 85% of battery charge even though I set my limit to 80%. Would upgrading to the pro version and using the discharge feature fix that issue?

Hi @Libroru,

We are currently not sure if this issue is only appearing in AlDente Free or also in AlDente Pro. However, with the Automatic Discharge feature of AlDente Pro, your MacBook would immediately discharge to the set charge limit if the battery percentage exceeds the charge limit. Furthermore, with the Sailing Mode active, the battery percentage is usually below the charge limit anyways.

Best,
Matthias

Libroru, 2023-10-10:
So I just bought AlDente Pro and I already restarted the app, but it seems like AlDente is not working at all!

I had my Macbook sit at 60% when I plugged it in and it now charged to 98% without the magsafe LED changing at all or the charging stopping / discharging being enabled.

Using on a Macbook M2 Air 
MacOS Sonoma

MatthiasKerbl, 2023-10-11:
> So I just bought AlDente Pro and I already restarted the app, but it seems like AlDente is not working at all!
>
> I had my Macbook sit at 60% when I plugged it in and it now charged to 98% without the [...]

Hi @Libroru,

Thank you for buying AlDente Pro, also, I am not sure I understand your exact issue.

Which charge limit did you set in AlDente?
Which setting did you choose in the Appearance/MagSafe LED section?
Do you have Automatic Discharge enabled?

Best,
Matthias

Libroru, 2023-10-11:
@MatthiasKerbl 

I set my charge limit to 80% and checked the `Control MagSafe LED` in the Appearance tab.
Automatic Discharge is indeed enabled.

Thanks for the quick answer!
I hope this makes things a bit clearer!

xezrunner, 2023-10-11:
I have the free version of the app. Since creating the issue, I haven't seen massive discrepancies.
I'm led to believe that it's simply macOS reporting the battery being lower than it is for a while.

[I now also see that Al Dente has a "hardware battery percentage" option in the Pro version.](https://apphousekitchen.com/feature-explanation-hardware-battery-percentage/)

MatthiasKerbl, 2023-10-11:
> I have the free version of the app. Since creating the issue, I haven't seen massive discrepancies. I'm led to believe that it's simply macOS reporting the battery being lower than it is for a [...]

Hi @xezrunner,

Thanks for getting back here and I am glad it works for you now as well.
I think we can close this issue now.

Best,
Matthias

closed 2023-10-11 by MatthiasKerbl