AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #923: AlDente Sailing Mode

TheGlorySaint, 2023-03-27:
**Describe the bug**
After I have purchased AlDente Pro, the Sailing Mode works no more. I have set the ChargeLimit to 80 Percent. The Sailing Mode is active and set to 20 Percent, so it should sail between 80 and 60 Percent, but it doesn't. I haven't changed any setting else.

I reinstalled, closed the app, force closed it over Activity Monitor and reinstalled the Helper, but nothing fixed the issue.

**To Reproduce**
Steps to reproduce the behavior:
Can't really tell.
I bought AlDente Pro and now sailing isn't working anymore. 

**Expected behavior**
Sailing should work

**Screenshots**

[image]
[image]
[image]
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook Pro 2021 16" M1 Pro 
 - macOS version Venture 13.2.1
 - AlDente Pro 
 - AlDente Version 1.21.3
 - List of all enabled/disabled settings in AlDente: See Screenshots

TheGlorySaint, 2023-03-28:
@MatthiasKerbl

Yesterday, I had plugged in my MacBook to a Thunderbolt dock to use multiple displays, from where the MacBook gets also the Power. It stayed the whole day at 80%, and it never dropped below. MacBook is in Clamshell mode.

MatthiasKerbl, 2023-03-29:
Hi @TheGlorySaint,

It really depends on these parameters (setting, MacBook model, charging setup, and usage behavior) when and how fast the battery level will drop. Especially, with the Apple Silicon MacBooks it can take a really long time before the battery level goes down. 

Cheers,
Matthias

TheGlorySaint, 2023-03-29:
@MatthiasKerbl

Thank you for your response.

I'm an App Developer. I sit around 10–12 Hours per Day in front of my MacBook. With that in mind, never seen a drop of percentage in Battery, doesn't seem right for me. But i'm fine with that. Keep Up the good work.

closed 2023-03-31 by MatthiasKerbl