AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #924: Aldente need password for helper every boot

rekby, 2023-03-28:
**Describe the bug**
When I reboot my macbook I need every time enter my password to aldente helper installer.

**Expected behavior**
No need password without aldente update.

**MacBook (please complete the following information):**
 - MacBook model (MacBook Pro, 16-inch, 2021)
 - macOS version (13.2.1 (22D68))
 - AlDente Pro
 - AlDente Version: 1.21.3
 - List of all enabled/disabled settings in AlDente:

**Additional context**
I have two partitions on my mac with two separated macos, aldente install in every of the partitions.

chepiok, 2023-03-28:
Same issue for me. Same hardware and software configuration.

papas68, 2023-03-28:
Same issue for me. Same hardware and software configuration.

MatthiasKerbl, 2023-03-29:
Hi everyone,

Thank you for reporting this. We are aware of this bug and working hard on fixing it.
Also, we already have a Github issue open for this bug here #866 where I will post updates about this bug. Therefore, I will close this issue.

Cheers,
Matthias

closed 2023-03-29 by MatthiasKerbl