AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #925: Schedule using which timezone?

DamonHD, 2023-03-30:
**Describe the bug**
I have set a Schedule task to happen at 15:58 (when local time was UTC), now it is happening at 16:58 (when local time is UTC+1).

**Expected behavior**
EIther follow local time and/or allow local time or an explicit timezone such as UTC.  Don't show 15:58 in schedule and run at 16:58!

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.
[image]

**MacBook (please complete the following information):**
 - MacBook model MacBook M1 Air 2020
 - macOS 13.2.1 
 - AlDente Pro
 - AlDente Version 1.21.3

DamonHD, 2023-03-31:
If you can without making things over complicated, allow the user to choose either:

1) Something which follows local time on the machine as displayed, eg including DST or even moving timezones when travelling.  That should probably be the default.

2) Something fixed to UTC (possibly with fixed offset).

Rgds

Damon

DamonHD, 2025-10-28:
There are exactly two times per year that it would be good to have fixed it by, for European users (I think that the UK is still in sync with the EU)!

DamonHD, 2026-10-06:
Please fix this before the clocks go back in the UK at the end of this month (October).

MatthiasKerbl, 2026-10-11:
Hi @DamonHD,

Thank you for your patience. We have fixed this, and the fix will be included in the next AlDente update, which we plan to release before the clocks go back at the end of October.

Best,
Matthias

MatthiasKerbl, 2026-10-19:
Hi @DamonHD,

The fix is included in AlDente 1.40 Beta 1: scheduled tasks now keep their time when the clocks change.

AlDente 1.40 Beta 1 is now available for AlDente Pro (Settings → gear icon at the bottom left → "Receive Beta Updates" → "Check for Updates"), and the final release for everyone will follow soon.

Best,
Matthias

DamonHD, 2026-10-19:
Hi,

Yes I have the beta loaded ready, thanks.

Damon