AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #926: Pro-functions stopped working

aerial-dot, 2023-04-01:
After the latest mac update (Mac OS Ventura 13.3 (22E252)) PRO functions of product stopped working, despite the fact that I purchased it few month ago.
Only free options available. When I click on "activate license" I can see the message that my pro version has already been registered (email and license code are also visible).
Tried to delete the app and install it - the problem still remain.
I also tried to deactivate the license in order to re-enter it, but the error appears.

MacBook Pro M1 2020 
AlDente Pro 
AlDente Version 1.21.3
Only free features available

2nthony, 2023-04-01:
I think it is because of macos 13.3, charge limit not working either.

limit to 80, but still fully charged

[image]

mvkirichenko, 2023-04-02:
exactly the same problem! After the latest mac update (Mac OS Ventura 13.3 (22E252)) PRO functions of product stopped working, despite the fact that I purchased it few month ago.
Only free options available. When I click on "activate license" I can see the message that my pro version has already been registered (email and license code are also visible).
Tried to delete the app and install it - the problem still remain.
I also tried to deactivate the license in order to re-enter it, but the error appears.

MacBook Pro M1 2020
AlDente Pro
AlDente Version 1.21.3
Only free features available

baf28, 2023-04-02:
The pro license is correctly activated only via VPN.

mbauer55, 2023-04-03:
Hey guys,

same here. After updating to 13.3, AlDente Pro stops working. 
I tried also calibration mode but in the step with the discharging it doesn't stop at ~75% it keeps discharging my MBP.

MacBook Pro M1 Max 2021
AlDente Pro 1.21.3

Thank u for checking.

mvkirichenko, 2023-04-04:
And support for those who purchased the pro version is a hoax? I wrote a letter to the developers but they just ignore me.

bugac, 2023-04-04:
same problem, 
macos 13.3 on mac air m1 (2020)

MatthiasKerbl, 2023-04-08:
Hi everyone,

Due to new sanctions, our third-party licensing partner had to restrict certain supported countries and territories. Therefore, activating and using AlDente Pro in some countries is only possible through a VPN service. You can read more about this issue and find a list of restricted countries and territories **[here](https://www.paddle.com/help/legal/sanctions/which-countries-are-supported-by-paddle)**.
We are very sorry for these inconveniences.

Kind regards,
Matthias

MatthiasKerbl, 2023-04-08:
> Hey guys,
> 
> same here. After updating to MacOS 13.3, AlDente Pro stops working. I tried also calibration mode but in the step with the discharging it doesn't stop at ~75% it keeps discharging my MBP.
> 
> MacBook Pro M1 Max 2021 AlDente Pro 1.21.3
> 
> Thank u for checking.

Hi @mbauer55,

This seems to be a different kind of issue and might be related to this issue: #768

Cheers,
Matthias

baf28, 2023-04-08:
Welcome to the world of pirates.

mvkirichenko, 2023-04-08:
developers, explain what will happen to the license for which I paid money?

timakden, 2023-04-09:
> Hi everyone,
>
> Due to new sanctions, our third-party licensing partner had to restrict certain supported countries and territories. Therefore, activating and using AlDente Pro in some countries is [...]

So I paid for the lifetime license of AlDente Pro and cannot use it because I was unlucky to born and reside in Russia? Really? What's the point of constantly checking of a **lifetime** license?

evdrug, 2023-04-17:
> Hi everyone,
>
> Due to new sanctions, our third-party licensing partner had to restrict certain supported countries and territories. Therefore, activating and using AlDente Pro in some countries is [...]

Hy @MatthiasKerbl !!!

The sanctions of your foreign partners apply to payments (new), but not as to those already made in the past. If we had not a lifetime license, but a subscription, yes, everything is OK! But with a lifetime license, third-party restrictions do not apply, this is a purchased product!!!

DamonHD, 2023-04-18:
Why be so rude?

This seems to be an international sanctions problem, as a side effect of people being killed in a war.

baf28, 2023-04-18:
> 

If we are talking about the war, then sanctions should have been imposed against those who started it and because of whom the war is going on and people are dying, and these are the countries of Europe and the United States. Why then do people from Russia and others who are fighting for freedom, peace and truth suffer?

baf28, 2023-04-18:
> This seems to be an international sanctions problem, as a side effect of people being killed in a war, who might have thought that they deserved a life.

If we are talking about the war, then sanctions should have been imposed against those who started it and because of whom the war is going on and people are dying, and these are the countries of Europe and the United States. Why then do people from Russia and others who are fighting for freedom, peace and truth suffer?

MatthiasKerbl, 2023-04-18:
Hi everyone,

Good news. The license server not working for restricted countries and territories turned out to be a bug and the sanctions only apply to payments in those regions. The issues with activating the licenses in these areas should be resolved now. Therefore, you can activate and use AlDente Pro without a VPN again.

We are sorry for the inconvenience.

Kind regards,
Matthias

closed 2023-08-03 by MatthiasKerbl