AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #930: Apple Studio Display is turned of when AlDente is changing the charging status

netzzwerk, 2023-04-07:
**Issue**
I own a M1 MacBook Air (Ventura 13.3) and a Studio Display (latest Software 16.4). As of today (I don't know what changed) the following behaviour started: The Battery is charged over 80% because of a reboot. I always use the Macbook in Clamshell mode. So Aldente starts to discharge the battery. With this status change (from holding the charge to discharging) the display turns of. With a press of a button the display turns back on. After a few second it turns of again. If I don't do anything the display stays of. If I repeatedly turn the display on again by pressing a keyboard button or my trackpad, Aldente crashes (disappears from the menu bar). After that the display stays on. If i start Aldente again, this cycle repeats.
The Problem does not occur when the Macbook is not used in clamshell mode but open. Then you can see that **AlDente is changing the charge-status from discharging to charging every few seconds**.
I also have MacBook Pro 13" M2 from my employer with macOS 12 Monterey. The error does not occur here. So this might be a Ventura Issue that has something to do with the new Studio-Display-Update 16.4

**To Reproduce**
Steps to reproduce the behavior:
1. let battery charge over the set limit
2. start aldente when Studio Display is connected
3. Wait for Aldente to discharge the battery
4. See the display shutting down.

**MacBook Air:**
 - MacBookAir10,1
 - Studio Display - OS: 16.4
 - macOS 13.3
 - AlDente Pro
 - AlDente 1.21.3

 **List of all enabled/disabled settings in AlDente:**
 - stop charging when sleeping: active
 - stop charging when AlDente is closed: not active
 - automatic discharge: active
 - deactivate sleep mode: not active
 - lock display when closed: not active
 - hardware battery percentage: not active
 - sailing mode: 5% active
 - heat protection max temo 45°: active

netzzwerk, 2023-04-08:
I just updated the MacBook Air to 13.3.1 that came out yesterday. Now I cannot recreate the issue. It seems like the OS-update fixed this issue. I will continue to monitor this issue. If this is reallly fixed I will close this in a few days.

netzzwerk, 2023-04-15:
This Issue seems to have been resolved by macOS 13.3.1
I close this issue.

stiebitz, 2023-05-23:
In the last few days, I've observed the same behavior under version 13.3.1 (a) with two different MacBook Pros M1/M1Max and one Studio Display. I've just updated them to version 13.4 and will report any new occurrences.

stiebitz, 2023-05-24:
Yes, this morning the first laptop showed this behavior under macOS 13.4 with the Studio Display attached.

MatthiasKerbl, 2025-07-15:
Hi everyone,

We have just released AlDente 1.34.1, which includes a fix for this issue. To get AlDente 1.34.1, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2025-07-15 by MatthiasKerbl