AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #931: No/very slow charging after Heat Protection turns off

stonerl, 2023-04-08:
**Describe the bug**
I was doing some CPU/GPU intensive work on battery. After the battery empty message appeared, I connected the MagSafe charger, but due to a high battery temperature, charging didn't start. After finishing my work, I disconnected the charger, I went to another room, connected the charger again and let my MacBook charge overnight (the lid was closed). The next morning, the MacBook's charge level was at around 20% and did not charge, although Aldente's Menubar Pop-up displayed a + and the MacSafe LED was orange, indicating that charging would be in progress. I had to unplug the charger and connect it again to successfully charge the MacBook. This happened 3 times so far.

**To Reproduce**
Steps to reproduce the behaviour:
1. Unplug the Charger
2. Play a game or do something that heats the system up, so Heat Protection kicks in
3. Battery empty warning appears 
4. Connect MagSafe
5. Continue working/playing
6. Finish working/playing
7. Close the Lid
8. Disconnect the charger
9. Connect the charger again.
10. After 5 h the percentage level is as 21% (or any other low value) and no charging happens
11. Disconnect Charger and connect again
12. Charging starts and charges to the set limit (e.g. 80%)

**Expected behavior**
The MacBook should charge to the set charge limit.

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro 14", 2023 – Apple M2 Pro
 - macOS version: 13.3
 - AlDente Pro (SetApp)
 - AlDente 1.21.3
 
 - List of all enabled settings in AlDente:

- Charge Limit 80%
- Stop charging when sleeping
- Automatic Discharge
- Disable Sleep until Charge Limit
- Hardware Battery Percentage
- Sailing Mode (5%)
- Heat Protection (35 C)
- Perecentage Icon
- Popup Animation
- Control Magsafe LED
- Task: Biweekly Battery Calibration
- Start Tasks at next Opportunity

MatthiasKerbl, 2023-04-08:
Hi @stonerl,

Thank you for the detailed explanation I think I know what is happening here.
After steps 7 and 8 the MacBook lid is closed and the MacBook is unplugged which means that the MacBook is asleep. Since you have "Stop charging when sleeping" enabled, your MacBook won't charge while it is asleep. This is why it stays at around 20% even when it is plugged in.
You can try to disable "Stop charging when sleeping", this should result in your MacBook charging in these situations but it might overcharge even to 100%. 

Cheers,
Matthias

stonerl, 2023-04-10:
Thanks for the swift reply. That makes sense. I now enabled disable sleep until charge limit. This should circumvent the problem as well, right?

closed 2023-04-10 by stonerl

MatthiasKerbl, 2023-04-10:
> Thanks for the swift reply. That makes sense. I now enabled disable sleep until charge limit. This should circumvent the problem as well, right?

Yes but only if you leave your MacBook plugged in. Otherwise, your MacBook will go to sleep.

Cheers,
Matthias