AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #935: Top Up does not work anymore after latest Macos Update

bitrecycling, 2023-04-12:
"Top Up" used wot work fine: press the button => charge limit is set to 100% => Macbook is charged to 100% 

Now after update to Macos Ventura 13.3.1 if I press "Top Up", charge limit is set 100%, Macbook is charging, but after a couple of seconds it hops back to the charge limit (80% in my case) and stops charging. 

Also weird and not observed before: the "sailing" now is little sequences of charging followed by discharging but on high frequency: changes multiple times per minute between each. (+ and pause symbols also switch).

bitrecycling, 2023-04-12:
Hardware: Macbook pro 16 2021 with Apple m1 pro.

keldey, 2023-04-14:
I don't have these issues on M2 and Ventura 13.3.1

MatthiasKerbl, 2023-04-15:
Hi @bitrecycling,

Your behavior might be triggered due to a known issue after macOS updates. You can read more about this issue and how to fix it in the thread here: #768

Cheers,
Matthias

bitrecycling, 2023-04-15:
@MatthiasKerbl after reading the whole other issue thread: how is this a duplicate? In my case "top up" does not work. The issue in #768 describes "charge limit isn't respected". Also the "fix" is described there is only for that problem ("discharge below charge limit"). Could you please investigate and supply a fix for my problem?

MatthiasKerbl, 2023-08-03:
Hi @bitrecycling,

Are you still experiencing these issues with our latest version AlDente 1.22.2?

Best,
Matthias

MatthiasKerbl, 2023-11-14:
Hi @bitrecycling,

Noting the extended period since our last communication, I am proceeding to close this issue for the time being. Please remember that should you need to revisit this matter, don't hesitate to reach out. I'll be more than happy to reopen the issue and assist you further.

Best,
Matthias

closed 2023-11-14 by MatthiasKerbl