AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #936: Purchased Lifetime License, but never received email/key

leungd, 2023-04-14:
Email used to purchase was me@derekleung.ca

MatthiasKerbl, 2023-04-15:
Hi @leungd,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Cheers,
Matthias

MatthiasKerbl, 2023-06-09:
Hi @leungd,

Noting the extended period since our last communication, I am proceeding to close this issue for the time being. Please remember that should you need to revisit this matter, don't hesitate to reach out. I'll be more than happy to reopen the issue and assist you further.

Best,
Matthias

closed 2023-06-09 by MatthiasKerbl