AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #938: (enhancement) Force sleep mode when battery below %

petervannes, 2023-04-15:
When on battery my Mac does not enter sleep mode regularly and drains the battery until hibernation is activated.
This is a long term issue i have not able to fix, there are no processes preventing sleep mode and I have experimented with different (pmset) configurations. But none of these settings do resolve this issue permanently.

Hereby a request to show a popup or notification when the battery level is below a certain percentage and the mac is running on battery. If the notification is ignored the Mac is forced into sleep mode. This prevents undesired battery draining and thus increasing the number of cycles.

MatthiasKerbl, 2023-04-15:
Hi @petervannes,

That sounds like an interesting idea, thank you for sharing.
In the short term, I would recommend checking out [Sleep Aid](https://ohanaware.com/sleepaid/). It has helped us to get more insights into sleep behavior of MacBooks.

Cheers,
Matthias

petervannes, 2023-05-13:
Thanks Matthias,

In the mean time i fixed this using the script below.

`# copy to /usr/local/bin and set permissions
# sudo cp sleeponit.zsh /usr/local/bin
# sudo chmod ugo+xr /usr/local/bin/sleeponit.zsh

powersource=$(pmset -g batt | head -n 1 | cut -c19- | rev | cut -c 2- | rev)
batteryPercentage=0
batteryLowLimit=55

# sleepNow 
function sleepNow {
  $(pmset sleepnow)
}

function onBattery {
  powersource=$(pmset -g batt | head -n 1 | cut -c19- | rev | cut -c 2- | rev)
 
  if [ "$powersource" != "AC Power" ]; then 
   return 0
  fi

   return 1
}

function batteryLevel {
  batteryPercentage=$(pmset -g batt | tail -n 1 | cut -c35- | cut -f 1 -d';' | tr -d "%" | bc) 
}

if (onBattery); then
  batteryLevel
  if [[ $batteryPercentage -le $batteryLowLimit ]]; then
    osascript -e "display notification \"Force sleep in 1 minute\" with title \"On battery power $batteryPercentage%\""
    sleep 60
    sleepNow
  fi
fi
`

`<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">

<plist version="1.0">
<dict>
    <key>Label</key>
    <string>SleepOnIt</string>
    <key>OnDemand</key>
    <false/>
    <key>ProgramArguments</key>
    <array>
        <string>/bin/zsh</string>
        <string>/usr/local/bin/sleeponit.zsh</string>
    </array>
    <key>KeepAlive</key>
    <true/>
    <key>RunAtLoad</key>
    <true/>
    <key>StartInterval</key>
    <integer>120</integer>
</dict>
</plist>
`

MatthiasKerbl, 2026-10-07:
Hi @petervannes,

Thank you for the suggestion and for sharing your script. We already have a GitHub thread open for low battery warnings and actions here: #323. I'll therefore close this issue to keep the discussion in one place.

Best,
Matthias

closed 2026-10-08 by MatthiasKerbl