AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #947: Calibration Mode

m-ali-awan, 2023-05-01:
I scheduled a Calibration Task, and it got triggered at set time. My charge limit was 70%, and it changed it to 100%. But now, it should discharge it to 15%, but I am monitoring it for around 7 hours now, but it is not discharging. Rather it is just keeping it at 100% and using Pwer Adapter as source.
[image]

MatthiasKerbl, 2023-08-03:
Hi @m-ali-awan,

Is the Calibration Mode still active within AlDente?

Best,
Matthias

MatthiasKerbl, 2023-11-14:
Hi @m-ali-awan,

Noting the extended period since our last communication, I am proceeding to close this issue for the time being. Please remember that should you need to revisit this matter, don't hesitate to reach out. I'll be more than happy to reopen the issue and assist you further.

Best,
Matthias

closed 2023-11-14 by MatthiasKerbl