AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #954: Losing settings after license key fails to re-validate

jkingan, 2023-05-17:
There are times when I am offline or in a restricted environment where AlDente is unable to refresh it's license. This usually happens after a reboot and then I get a message that I need to connect to the internet and let it do it's thing. After restoring my internet connection and then restarting AlDente, it does it's license dance and shows I have pro again. However, all my settings are now set back to defaults and I need to redo them. Sailing configuration, icon settings - everything is reset.

Steps to reproduce the behavior:
1. Wait for AlDente to need to check it's license
2. Disconnect from the internet
3. Reboot
4. See warning from AlDente about needing to validate every 30 days
5. Quit AlDente
6. Reconnect to the internet
7. Start AlDente, let it verify the license
8. Verify all your settings are now reset (I think this actually happens back in step 4)

AlDente Pro 1.21.4

I've been unlucky enough to have this happen three times. After the second time I copied all the settings down in a text file so I could get them back where I had them.

MatthiasKerbl, 2023-05-18:
Hi @jkingan,

thank you for sharing this issue and I am sorry for the inconvenience. We are already having this issue on our backlog and will fix it with one of the next updates.

Cheers,
Matthias

MatthiasKerbl, 2023-06-15:
Hi @jkingan  ,

We have just released AlDente 1.22 which includes a fix for this issue. To get AlDente 1.22 you just need to update within the app under Settings/General/Check for Updates.

Best,
 Matthias

closed 2023-06-15 by MatthiasKerbl