AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #955: Having Charge Limit at 70%, battery would ocasionally hit 71% and discharge for a couple of seconds

vladandrei51, 2023-05-21:
Pretty much the issue in the title.

This is especially annoying because it messes with the brightness when the discharge occurs. I didn't have this issue in the past.

Other possible relevant info:
Hardware Battery percentage on
Sailing mode 5% on
Heat Protection on/off makes no difference
Using AlDente 1.21.4

MatthiasKerbl, 2023-06-09:
Hi @vladandrei51,

This is most likely caused by fluctuations in the battery percentage. When the battery percentage "jumps" to 71% due to the margin of error in the battery percentage, Automatic Discharge kicks in and discharges your MacBook down to 70% again. Unfortunately, there is not much that can be done to avoid this behavior. You could try to run the Calibration Mode to increase the accuracy of the battery percentage. If this issue still persists, consider turning off the Automatic Discharge feature.

Best,
Matthias

MatthiasKerbl, 2023-11-14:
Hi @vladandrei51,

Noting the extended period since our last communication, I am proceeding to close this issue for the time being. Please remember that should you need to revisit this matter, don't hesitate to reach out. I'll be more than happy to reopen the issue and assist you further.

Best,
Matthias

closed 2023-11-14 by MatthiasKerbl