AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #958: Error in Limiter function and charge and discharge frequently after a while of using, and sailing mode is not working.

Aghyads, 2023-05-30:
**Dear developer, since a week/10 days, my macbook charge and discharge automatically, so every 5-10 minutes I hear the charging tone which was not the case earlier.
Also, the sailing mode is not working, eventho it is ON. the laptop is charging or discharging based on my inputs, not automatically as it was in the first period.
Also, I got many times per day the limiter error, so I need to turn off then on again**

**MacBook (please complete the following information):**
 - MACBOOK Pro 2019
 - Ventura 13.4
 - AlDente Pro
 - AlDente Version: 1.21.4
 - List of all enabled/disabled settings in AlDente:
 - Sailing, heat protection, Calibrated before once, animation, pop up, icon and percentage of battery, stop charge when sleep, stop charge when turned off, automatic discharge, hardware battery. 

**Additional context**
I didn't had this issues like two weeks ago.

I would like to hear back from you regarding potential solutions please. 

Thank you 🙏
Regards, Aghyad

bc0ss, 2023-05-31:
I can confirm a very similar issue started for me after upgrading to 13.4 (ventura).  

It starts out limiting the charge (plug icon on battery), but after about 30s goes back to charging the battery (lightening bolt on battery).  

I use AlDente to force my dell display to work properly with my macbook by preventing the macbook from charging.  When the macbook begins to charge like this the display turns off.

Aghyads, 2023-06-10:
@MatthiasKerbl dear any updates over here?
two weeks already and still facing the same issue (premium version). Sealing is not working, I should always decrease or top up manually, or hold the battery at a certain percentage. So the app is not functioning correctly. 
Would you please check the bug and inform us of possible solutions? 
Thanks in advance!

MatthiasKerbl, 2023-06-10:
Hi @Aghyads,

Sorry for the late reply.

Regarding automatic charging and discharging, this might be due to fluctuations in your battery percentage in combination with Automatic Discharge. Try to disable the Automatic Discharge feature, afterward, you should not experience this issue anymore.

Regarding the Sailing Mode, it is a common misconception that the Sailing Mode does actively charge and discharge your MacBook between the sailing range. This would actually be much worse for the battery.
You can read a detailed description of what the Sailing Mode actually does on our blog "[Feature Explanation: Sailing Mode](https://apphousekitchen.com/feature-explanation-sailing-mode/)"

Regarding the "limiter error" notification, we are releasing a new update next week which will solve this issue.

Best,
Matthias

closed 2023-10-11 by MatthiasKerbl