AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #959: iOS Companion App

stonerl, 2023-05-30:
**Describe the solution you'd like**
I wonder if it would be possible to have a companion app for iOS/iPadOS to remotely control AlDente. One possible example would be: Remotely enable Top Up or Calibration Mode, or change the charging limit.

The reasoning for this would be that I usually have my MacBook connected to a docking station, and sometimes need to Top Up the battery, when I'm not in my office, e.g. during a meeting. With such an app, I could tell AlDente to charge to a 100%, pick the MacBook up later and have a fully charged battery on the go.

MatthiasKerbl, 2023-05-30:
Hi @stonerl,

Awesome idea. However, it might be difficult to implement since your MacBook would mostly be in sleep but we will put it on our list for potential future improvements.

Cheers,
Matthias

stonerl, 2023-05-30:
I guess to make this work, these system settings should be applied by the user:

- Put hard disks to sleep → Only on Battery / Never
- Wake for Network Access → Only on Power Adapter / Always

[image]

Another way would be to have an option in AlDente like:

[ ] Enable remote access/companion app

and when enabled, a user could additionally select the following:

[ ] Disable Sleep Mode while on Power Adapter