AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #960: Sailing Mode does not work as expected

sergiupopescu199, 2023-06-02:
**Describe the bug**
Sailing mode doe snot work as expected: I was expecting for Sailing mode to work after it charge the battery to 55% it should go to 45% & recharge again to 55% & so on but it stays on 55% all the time

[image]

[image]

[image]
=

**MacBook (please complete the following information):**
 - MacBook model 13 inch M2 16GB
 - macOS version Ventura 13.4
 - AlDente Pro
 - AlDente Version -> 1.21.4
 - List of all enabled/disabled settings in AlDente:

MatthiasKerbl, 2023-06-09:
Hi @sergiupopescu199,

This is actually the intended behavior of the Sailing Mode and looking at your screenshots, Sailing Mode is working as intended.
It is a common misconception that the Sailing Mode does actively charge and discharge your MacBook between the sailing range. This would actually be much worse for the battery.
You can read a detailed description of what the Sailing Mode actually does on our blog "[Feature Explanation: Sailing Mode](https://apphousekitchen.com/feature-explanation-sailing-mode/)"

Best,
Matthias

sergiupopescu199, 2023-06-09:
@MatthiasKerbl Thank you for detailed explanation now I understood!

closed 2023-06-09 by sergiupopescu199