AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #961: Menubar: Window does not disappear when focussing other menu items

tisba, 2023-06-03:
**Describe the bug**
Window (the one that pops-up after you click the menu icon) does not disappear when focussing other menu items. This is quite annoying and also inconsistent with every other macOS app that uses the menubar in this way.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on AlDente menu item
1. Click on another menu item
1. AlDente window is still open and overlapping with others

**Expected behavior**
Like all other apps, the overlay/window should disappear.

**~Screenshots~Video**

https://github.com/davidwernhart/AlDente-Charge-Limiter/assets/75156/4cf30879-9513-4985-a9d1-b36d19c73cac

**MacBook (please complete the following information):**
 - MacBook model: MacBookPro18,3 (14", M1 Pro)
 - macOS version: 13.4
 - AlDente Pro
 - AlDente Version: 1.21.4
 - List of all enabled/disabled settings in AlDente:

[image]

tisba, 2023-06-04:
Actually, the window does not appear also when a different window gets focus. Only clicking on the menu item again hides the overlay.

MatthiasKerbl, 2023-06-09:
Hi @tisba,

Thank you for this thorough bug report, we will fix it with our next update.

Best,
Matthias

MatthiasKerbl, 2023-06-15:
Hi @tisba ,

We have just released AlDente 1.22 which includes a fix for this issue. To get AlDente 1.22 you just need to update within the app under Settings/General/Check for Updates.

Best,
 Matthias

closed 2023-06-15 by MatthiasKerbl

tisba, 2023-06-16:
Great, thanks a lot for the quick fix!