AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #962: Letter C in brackets used instead of Copyright symbol

ADTC, 2023-06-04:
Just installed and trying out the app. I noticed the main settings window has `(C)` (i.e. uppercase letter C in brackets) instead of the copyright symbol, while the About dialog has the correct symbol. I thought it's perhaps missed out on. 🙂 

[image]

MatthiasKerbl, 2023-06-15:
Hi @ADTC,

Thanks again for letting us know. It is fixed in our newest version now.

Best,
Matthias

ADTC, 2023-06-28:
Hmm thank you for fixing. Although, I'm on 1.22 and it's still the same (not fixed) 😆

[image]

Maybe the next version?

MatthiasKerbl, 2023-06-29:
> Hmm thank you for fixing. Although, I'm on 1.22 and it's still the same (not fixed) 😆
> 
> [image]
> Maybe the next version?

Hi @ADTC,

You are right, we missed it in the Free version. Will be fixed with the next update.

Best,
Matthias

MatthiasKerbl, 2023-08-03:
Hi @ADTC,

This is fixed now with our newest update.

Best,
Matthias

closed 2023-08-03 by MatthiasKerbl