AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #963: Crash by click into dropdown icon

masterzone3d, 2023-06-05:
**Describe the bug**
Every time I click into the little arrow where the task history is stored, this symbol: ^ the panel closes and crashes....

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ' the al dente panel'
2. Click on 'schedule button'
3. Scroll down to 'task history  '
4. click to ^ icon for collapse or compact and the app crashes..

**Expected behavior**
Mu Ardent has about four or five lines wrote into the task history

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.
[image]

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro M1 Max
 - macOS version : Ventura 13.4 (22F66)
 - AlDente Pro 
 - AlDente Version: 1.21.4

https://github.com/davidwernhart/AlDente-Charge-Limiter/assets/132667740/53a5cf76-0754-47fc-a787-d01329d13109

MatthiasKerbl, 2023-06-09:
Hi @masterzone3d,

Thank you for the detailed report and I am sorry you are experiencing this issue.

Unfortunately, we are currently not able to reproduce this issue. However, one potential fix for your issue could be to reset the settings within AlDente. Open the settings window of AlDente, go to "General" and click the "Reset Settings" button in the lower right corner.
Please let us know if this resolved the issue.

Best,
Matthias

masterzone3d, 2023-06-12:
thanks Matthias for your time. I tried to reset settings so I will continue to take a deep look into when some calibrations log will be written into the GUI area..
Thanks...

MatthiasKerbl, 2023-08-03:
Hi @masterzone3d,

Are you still experiencing this issue with our newest version AlDente 1.22.2?

Best,
Matthias

masterzone3d, 2023-08-04:
Dear Matthias,

Sorry for my later replay and nice to read you again and also greetings for your amazing app, it saves my MacBook Pro battery!
I tried to click the arrow that collapse the task history but seems fine for now…I would like to try later for test it when there is more text inside the frame…
Anyway right now seems works smoothly…
See you
Alex :)

| Alessandro Maniscalco
| www.c4dzone.com
| Maxon Training Center

| www.zuccherodikanna.com
| Produzioni in computer grafica 3D
| Via Mezzacampagna 4
| 37135 Verona
| Cell aziendale: ‭+39 351 6097868‬

closed 2023-08-04 by MatthiasKerbl

masterzone3d, 2023-09-12:
> Hi @masterzone3d <https://github.com/masterzone3d>,
>
> That's good to hear, I will close this issue for now but will open it again if anything comes up.
>
> Best,
> Matthias

Dear Matthias, I would like to write you about the crash issue described time ago. I tried to collapse and open again the area where taks history is placed.
And boom, the crash occurs…
Let me know If you need some info for investing…anyway is not a big problem, maybe I’m the only one that try to compact and collapse with the  ^ symbol so is not annoying anymore…
Thanks
Alex

￼

| Alessandro Maniscalco
| www.c4dzone.com
| Maxon Training Center

| www.zuccherodikanna.com
| Produzioni in computer grafica 3D
| Via Mezzacampagna 4
| 37135 Verona
| Cell aziendale: ‭+39 351 6097868‬