AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #966: Request: Firmware based charge limit on Apple Silicon Macs

CMMChris, 2023-06-11:
**Is your feature request related to a problem? Please describe.**
Currently, AlDente controls the charge limit only when running on Apple Silicon Macs, which can cause the charging limit being exceeded in some cases, for example during an OS update or reboot, where the device starts charging once AlDente is being killed or sometimes it even happens in sleep.

**Describe the solution you'd like**
According to the Asahi Linux developers, the latest Apple Silicon Firmware versions starting from macOS 13 offer support for charge limit even when no OS is booted, so the charging is halted even when you reboot. They are actually working on making use of this mechanism. It does have a hardcoded 80% limit tho, but it still would be nice if AlDente could utilize the firmware based charge limit as well with some future update. This way, while a charge limit could still be exceeded, it at least would not go above 80%.

https://www.reddit.com/r/AsahiLinux/comments/13g2h3b/comment/jk3odu1/?utm_source=share&utm_medium=web2x&context=3

MatthiasKerbl, 2023-06-13:
Hi @CMMChris,

Thank you for the hint, we are always looking into new possibilities to improve AlDente.
Are you sure that it also works during OS updates and reboots? Because from what we could see, they were using the same technical pathways we are using and they only work while macOS is running or in sleep mode.

Best,
Matthias

CMMChris, 2023-06-14:
No, there was a new charge limit feature that Apple added to its firmware with macOS 13, that when enabled enforces a fixed 80% limit. It's probably used by macOSes own charging control in order to prevent charging to 100% when the machine is shut down (just my assumption). It was also mentioned here when they added support for it in Asahi: https://social.treehouse.systems/@AsahiLinux/110497512340479064

MatthiasKerbl, 2023-06-14:
> No, there was a new charge limit feature that Apple added to its firmware with macOS 13, that when enabled enforces a fixed 80% limit. It's probably used by macOSes own charging control in order to [...]

Hi @CMMChris,

Thanks again, however, from what I can read in this thread, it only works during sleep. Nothing was mentioned about the behavior when the MacBook is shut down.

> "This new charge control method also works in sleep mode."

Best,
Matthias

CMMChris, 2023-06-14:
From what I understood, it is firmware controlled and thus independent of the OS once triggered. Maybe you can get in touch with Hector Martin to clarify. Unfortunately, I cannot test this at the moment since the M2 Max isn't fully supported by Asahi Linux yet and the macOS optimized battery charging is never triggered by my usage behavior.

double-thinker, 2023-08-14:
Thanks for the hint @CMMChris after some research I developed [a preliminary implementation](https://github.com/actuallymentor/battery/pull/163). Use with caution, not very tested yet but seems to work fine). Maybe this is useful for AlDente team. Based on [Asahi's solution](https://github.com/AsahiLinux/linux/commit/107ed86e21d1522d5d5d9d629f1d9c371e37df7f#diff-6b2c1b57b60e2829a9973ea707f596a3ff85fa787eca5d6d3b8ac79ebdd0381cR453)

CMMChris, 2023-08-14:
Nice, hopefully this will be implemented in AlDente as well.

sieren, 2023-11-21:
Would be great to see this in AlDente indeed, however I suppose more research is needed.

MatthiasKerbl, 2024-11-21:
Hi everyone,

Since Apple removed this possibility of manipulating the charging behavior, I will close this issue.

Best,
Matthias

closed 2024-11-21 by MatthiasKerbl