AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #973: Power Flow MB 2017 Four TB not working

Oll0w4in, 2023-06-17:
**Describe the bug**
Since the Power Flow feature is in beta, I would like to report a bug on my MB Pro 2017 with four TB Ports. The new Power Flow feature is not showing anything at all. I will append an image of the stats site.

Let me know how I can help you with this bug.

**To Reproduce**

1. Install newest Version of AlDente
2. Activate Power Flow feature
3. Try to display the new feature in the app or in the menu bar

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook model: MacBookPro14,2
 - macOS version 13.4 (22F66)
 - AlDente Pro
 - AlDente Version 1.22
 - List of all enabled/disabled settings in AlDente:
 - Start at login enabled
 - stop charging when sleep and turned off enabled
 - auto discharge enabled
 - hardware battery percent enabled
 - sailing enabled, 10% from 65%
 - heat protection enabled at 38°C
 - Show Popover enabled

Rennleiter02, 2023-07-06:
Exactly the same problem (on identical model, macOS 13.4.1 (22F82))

MatthiasKerbl, 2023-07-19:
Hi @Oll0w4in,

Thanks for letting us know, we will release a fixed version of AlDente which supports your model soon.
Further updates about this will be provided here #985.

Best,
Matthias

closed 2023-07-19 by MatthiasKerbl