AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #974: Menubar item cut off

stonerl, 2023-06-17:
**Describe the bug**
When setting the Menubar Icon to “Show percentage”, the percentage (%) symbol is cut off when the battery is charged to 100%. See the attached screenshot.

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook Pro 14"
 - macOS 13.4
 - AlDente Pro
 - AlDente Version 1.22

MatthiasKerbl, 2023-06-17:
Hi @stonerl,

Thanks for making us aware of this bug, we will fix it with the next update.

Best,
Matthias

MatthiasKerbl, 2023-08-03:
Hi @stonerl,

This bug got fixed with our latest update.

Best,
Matthias

closed 2023-08-03 by MatthiasKerbl