AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #975: Sailing mode does not work after update to 1.22

Fonliar, 2023-06-18:
Hello,

after the update to 1.22 the sailing mode does not work. 
Neither the LED goes to green and the MacBook still charges. 
My sailing range is 20 to 80% and now the battery is at 67% and 
when I plug in the cable it starts with charging. 
I installed the helper new and switched the sailing mode on and off and 
changes the maximum charge but nothing works for me. 
Any ideas or solutions?

MacBook Air M2 2022
Ventura 13.4
AlDente Pro
1.22

[image]

[image]

[image]

MatthiasKerbl, 2023-06-18:
Hi @Fonliar,

The Sailing Mode always starts from the upper limit. Therefore, try to charge to the set charge limit and when the charge limit is reached, check if you can see the little "paper plane" icon in the slider of the popup window.
If you want to know more about the Sailing Mode, check out our [article on our blog](https://apphousekitchen.com/feature-explanation-sailing-mode/)

[image]

Best,
Matthias

kamenskiy24, 2023-06-23:
Hey,

same thing here. I turn off my MacBook every night. And when I turn it on the next day it starts charging to the upper limit of the sailing mode. This behavior was introduced in the latest update. I don't need my MacBook to be charged to the upper limit on plugging it in. My sailing mode is 70% - 60%. When I turn it on it starts charging from 68% to 70%. I believe this wears out my battery.

yfzhou0904, 2023-06-23:
> Hi @Fonliar,
>
> The Sailing Mode always starts from the upper limit. Therefore, try to charge to the set charge limit and when the charge limit is reached, check if you can see the little "paper [...]

Hi, I am observing the same problem after upgrading to v1.22.
I think @Fonliar knows exactly how sailing mode is supposed to work, but it is not working on my machine either. I set my charge limit at 80% with a 50% sailing difference, meaning the battery shouldn't start charging unless it goes under 30%, yet it charges at 70%. I'm on a 14 inch M1 MBP.

[image]
[image]

l4vem, 2023-06-24:
Hi all, I had the same problem. I uninstalled and reinstalled the helper application. That didn't work. I then changed the sailing mode percentage from 10% to 15% and that fixed it. Changed it back to 10% and now working as expected.

Fonliar, 2023-06-29:
With 1.21.4 now I have the same problems. Sailing mode does not work. 

MacBook is at 65% battery, charge limit is set to 80%, sailing is activated and an interval of 60% is set. 
When I plug in the Charger, the MacBook starts charing, but it should sail to/till 20%... Like in the past.

MatthiasKerbl, 2023-06-29:
Hi everyone,

Thank you all for sharing your experience with this issue. We are currently investigating this issue and will give updates about it here.

Best,
Matthias

natebot88, 2023-07-10:
Hey all, have you tried disabling the Optimized Battery Charging in System Settings?
System Settings > Battery> Battery Health (click the "i" icon). This built-in feature was messing with my Sailing Mode. All is working perfectly now that it is disabled.

MatthiasKerbl, 2023-07-19:
Hi everyone,

We are quiet sure to have found the cause of this issue and will fix it with the next update soon. :)

Best,
Matthias

FiftyFathom, 2023-07-23:
> I tried it on my other MacBook (M1 Max) and there is the same problem. It does not depend on the M2 chip. @MatthiasKerbl I use sailing mode works many months without any issues and I know how the sailing mode works ;)
> 
> Any updates? On the MacBook Air (M2) from my wife with 1.21 the sailing mode works fine. It has the same specs like my M2 Air but another colour...
> 
> I am going back to 1.21.4.

Where can I find previous version as I also have the same issue.

shahin365, 2023-07-26:
> Hi everyone,
> 
> We are quiet sure to have found the cause of this issue and will fix it with the next update soon. :)
> 
> Best, Matthias

Hey, I have the same problem. any update?

MatthiasKerbl, 2023-07-27:
> > Hi everyone,
> > We are quiet sure to have found the cause of this issue and will fix it with the next update soon. :)
> > Best, Matthias
> 
> Hey, I have the same problem. any update?

Hi @shahin365,

We are in the final stages of testing the update and will release it soon.

Best,
Matthias

kvaellning, 2023-07-31:
On my M1 MacBook Pro, this issue is not fixed in 1.22.2.

FiftyFathom, 2023-08-01:
Same here on M2 Pro, sailing mode still not working on 1.22.2

MatthiasKerbl, 2023-08-01:
Hi @kvaellning and @FiftyFathom,

The Sailing Mode does now continue working after restarting my Apple Silicon MacBook.
What is the exact issue you are encountering, could you provide a step-by-step guide to reproduce your issue?

Best,
Matthias

kvaellning, 2023-08-01:
@MatthiasKerbl 
Okay, I'll try my best (but maybe the wording around the sailing mode just needs more clearup):

Scenario A:
- Limit: 70% | Difference: 50% <br> --> Sailing Mode interval [20...70] %
- MacBook starts at 70%, stays plugged, battery drains slowly
- Sailing Mode appears to work (but since battery depletion is really low, maybe this is a misconsception)

Scenario B:
- Limit: 70% | Difference: 50% <br> --> Sailing Mode interval [20...70] %
- MacBook starts at 70%
- MacBook gets unplugged and used some time unplugged. New Level: 60%
- MacBook replugged to power supply. Battery starts to get charged.

This is at least (for me) unexpected behavior, since 60% is inside the defined interval [20...70] %.

Understanding based on the documentation:
- Limit: 70% | Difference: 50% <br> --> Sailing Mode interval [20...70] %
- Battery Level: 10%
- MacBook is plugged to power supply, starts loading until reaching upper level of sailing mode (limit)

My current workaround is to set the limit to my minimum value sailing mode value and leave it there. From my understanding of the documentation that is not intended behaviour :)

FiftyFathom, 2023-08-01:
After installing the 1.22.2, I also re-installed the helper tool and then did a restart. My settings are set to 70% charge limit with sailing mode set to 20% difference, and stop charge when app is closed, max temp set to 35c. Optimized battery off from the settings on the macbook.
Macbook while plugged in, when reaches 70% of charge, charge is stopped, but it never drops, always stays at 70%, it does not sail down to 50% which is 20% difference. Even after days on plugged in the % is always at 70% which is my selected limit.

MatthiasKerbl, 2023-08-01:
> @MatthiasKerbl Okay, I'll try my best (but maybe the wording around the sailing mode just needs more clearup):
> 
> **Scenario A:**
> 
> * Limit: 70% | Difference: 50%  --> Sailing Mode interval [20...70] %
> * MacBook starts at 70%, stays plugged, battery drains slowly
> * Sailing Mode appears to work (but since battery depletion is really low, maybe this is a misconsception)
> 
> **Scenario B:**
> 
> * Limit: 70% | Difference: 50%  --> Sailing Mode interval [20...70] %
> * MacBook starts at 70%
> * MacBook gets unplugged and used some time unplugged. New Level: 60%
> * MacBook replugged to power supply. Battery starts to get charged.
> 
> This is at least (for me) unexpected behavior, since 60% is inside the defined interval [20...70] %.
> 
> **Understanding based on the documentation:**
> 
> * Limit: 70% | Difference: 50%  --> Sailing Mode interval [20...70] %
> * Battery Level: 10%
> * MacBook is plugged to power supply, starts loading until reaching upper level of sailing mode (limit)
> 
> My current workaround is to set the limit to my minimum value sailing mode value and leave it there. From my understanding of the documentation that is not intended behaviour :)
> 
> Cheers, Juliane

Hi Juliane,

Thank you for the detailed report.
Just to clarify, the Sailing Mode always starts from the upper limit and is active when you see the little paper plane icon within the slider bar. Can you confirm that in both scenarios, you started at the upper limit with the Sailing Mode active as indicated on the picture below?
[image]

Best,
Matthias

MatthiasKerbl, 2023-08-01:
> After installing the 1.22.2, I also re-installed the helper tool and then did a restart. My settings are set to 70% charge limit with sailing mode set to 20% difference, and stop charge when app is [...]

Hi @FiftyFathom,

I think there is a misunderstanding about what the Sailing Mode actually does. Just to clarify, the Sailing Mode does not discharge the battery within the sailing range. The battery only discharges due to the following three known reasons: 1. Self-discharge over time (very slowly), 2. the battery jumps in as a buffer for power spikes (very slowly), and 3. when the MacBook is unplugged and used without being connected to the charger. Depending on your MacBook model and usage behavior, the battery percentage might drop very slowly or even never. You can read more about this on our blog [Feature Explanation: Sailing Mode](https://apphousekitchen.com/feature-explanation-sailing-mode/).

Best,
Matthias

kvaellning, 2023-08-01:
Dear @MatthiasKerbl,
I'll take a look on that in the afternoon. I assume that the sailing mode does not come to life, for whatever reason this could be.

From yesterday's tests, the interval was set to [20...70] % and my battery was @48% (hardware battery level, OSX level roughly the same). For whatever reason, this lead the sailing mode to not kick in, and the battery was charged.
If I can provide you with any logs which may help, feel free to tell me how :)

Cheers,
Juliane

kvaellning, 2023-08-01:
So, it looks like that I don't reach sailing mode:
[image]

With 38% I should be afaik in the interval of the sailing mode.

In case that it helps, I have also attached the debug data:
[aldentepro-debug-2023-08-01-16:00:51.pasta.zip](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/12230023/aldentepro-debug-2023-08-01-16.00.51.pasta.zip)

Cheers,
Juliane

MatthiasKerbl, 2023-08-01:
> So, it looks like that I don't reach sailing mode: [image]
>
> With 38% I should be afaik in the interval of the sailing mode.
>
> In case that it helps, I have also attached the debug data: [...]

Hi Juliane,

Correct, the Sailing Mode is not (yet) active in your case. You just need to let it charge to your upper limit of 70%, afterward, the Sailing Mode should be active.

Best,
Matthias

kvaellning, 2023-08-01:
> Hi Juliane,
> 
> Correct, the Sailing Mode is not (yet) active in your case. You just need to let it charge to your upper limit of 70%, afterward, the Sailing Mode should be active.
> 
> Best, Matthias

Okay, apparently I misinterpreted example 3 :)
So then it works as explained.

cheers,
Juliane

kvaellning, 2023-08-02:
Hej @MatthiasKerbl,

I now ran in the following problem:
- charged until upper limit (sailing mode gets activated)
- cutting the power supply and playing around a bit to decrease the battery level
- restarting
- plugin power supply
- sailing mode is inactive

I understand that there are limitations regarding Apple Silicon, but this was a little bit unexpected to me. I assumed the sailing mode state will be stored somewhere persistently (e.g. HDD) and, if necessary restored after reboot (so that I need to enter it at least once to activate the feature, and then 'it just works').

cheers,
Juliane

shahin365, 2023-08-02:
> So, it looks like that I don't reach sailing mode: [image]
>
> With 38% I should be afaik in the interval of the sailing mode.
>
> In case that it helps, I have also attached the debug data: [...]

unrelated to the topic. can I ask what weather app you use? (the icon on the menu bar). thanks.

kvaellning, 2023-08-02:
> unrelated to the topic. can I ask what weather app you use? (the icon on the menu bar). thanks.

Actually, it's an app to read out the temperature sensors (called Hot).

MatthiasKerbl, 2023-08-03:
> Hej @MatthiasKerbl,
>
> I now ran in the following problem:
>
> * charged until upper limit (sailing mode gets activated)
> * cutting the power supply and playing around a bit to decrease the battery level [...]

Hi @kvaellning,

This scenario works without any issues for me (Apple Silicon MacBook).

- Charged up to charge limit to activate Sailing Mode.
- Unplug the charger.
- Drain the battery within the set sailing range.
- Restart MacBook.
- Plug in the charger again.
- Sailing Mode stays active and the battery is not charging.

Which MacBook model do you have?
Which macOS version do you use?

Best,
Matthias

MatthiasKerbl, 2023-08-03:
> The issue unfortunately remains unresolved for me with 1.22.2. Sailing mode still doesn't persist after a reboot as it did before 1.22.
> 
> **Behavior:**
> 
> 1. As long as the system doesn't reboot Sailing mode works correctly. When my charge limit of 70% is reached the charge is paused. If the charge drops within the Sailing mode range it remains paused with the Sailing mode icon visible.
> 2. However, if I reboot the system with Sailing mode active it does not persist after the reboot and Al Dente allows the system to charge back up to its charge limit.
> 
> The behavior before 1.22 was that Sailing mode persisted and charging would remain paused after a reboot until the charge drops below the difference set in Sailing mode.
> 
> **My upgrade steps:**
> 
> 1. Removed the helper application from settings.
> 2. Uninstalled Al Dente with Homebrew.
> 3. Re-installed Al dente with Homebrew.
> 4. Re-installed the helper script.
> 
> **Specs:** 16" MacBook Pro M1 Pro Ventura 13.5 AlDente Pro 1.22.2

Hi @Freddy500,

This scenario works consistently without any issues on our Apple Silicon MacBook and other users have also already confirmed that it works for them.

Could you send a detailed step-by-step description of what exactly your behavior is?

Best,
Matthias

kvaellning, 2023-08-03:
@MatthiasKerbl 
Okay, two questions:
- it doesn't matter if I restart, or turn off and on again, right? (I work with embedded stuff, so there are differences there)
- my usual routine is to unplug the charger, turn off the Macbook (go to sleep and work :D ) and then start the Macbook and turn on the power after I logged in. FWIK sailing mode should work here.

I add the MacBook/macOS version here via edit when I have access to my computer.

Cheers,
Juliane

Freddy500, 2023-08-03:
> Hi @Freddy500,
> 
> This scenario works consistently without any issues on our Apple Silicon MacBook and other users have also already confirmed that it works for them.
> 
> Could you send a detailed step-by-step description of what exactly your behavior is?
> 
> Best, Matthias

Hi Matthias,

I've done a bit of testing and the issue seems related to how much I let the battery drain after my charge limit is reached and Sailing mode is activated. In my case there seems to be a threshold for a discharge that is too small before Sailing mode persists after a reboot. 

**My settings:**
- Charge limit set to 70%.
- Sailing mode set to a 20% difference.
- Hardware Battery Percentage enabled.

**Scenario 1: Sailing mode persists**
I unplug the system and let the battery discharge to 67% or lower (70% as reported by MacOS). I reboot, plug the system back in and the charge is paused at the same 67%.

**Scenario 2: Sailing Mode does not persist**
Same as above but I discharge only to 68% or 69% hardware battery percentage (= 71% or 72% for MacOS). Such small discharges can happen after a day of working plugged in. After a reboot and plugging the system in, it charges back up to 70%.

The cutoff point for me seems to be a discharge of 2%. I mention the discrepancy between the hardware battery percentage I have enabled and what MacOS reports as the difference amounts to about the same threshold where the cutoff happens.

Thanks,
Freddy.

MatthiasKerbl, 2023-08-03:
> @MatthiasKerbl Okay, two questions:
>
> * it doesn't matter if I restart, or turn off and on again, right? (I work with embedded stuff, so there are differences there)
> * my usual routine is to unplug [...]

Hi @kvaellning,

It does not matter if you restart or turn off and on the MacBook manually. However, in order for all of this to work, it is important to unplug the MacBook while macOS and AlDente are still running and plug it in again when macOS and AlDente are running again.

Best,
Matthias

Freddy500, 2023-08-03:
> Hi Freddy,
> 
> Let me try to explain it in general. The current battery percentage (State of Charge) is always only a good guess and never completely accurate. Therefore, the SoC likes to "jump around" by let's say 2% percent sometimes anyways. On top of that, due to limitations from Apple, AlDente can not prohibit charging at all when the MacBook is turned off or even during the restart process (on Apple Silicon MacBooks). This means, during the restart process and especially when you manually turn it off and on, the MacBook will charge the Battery for a while. Depending on your MacBook model and charger, this could mean that your Battery gets charged between 0 and a couple of percent. This means in summary that you should always test these behaviors with at least a 5% margin to counteract the margin of error and mechanisms in place mentioned above.
> 
> Best, Matthias

Hi Matthias,

Thank you for your answer.

I believe I understand what you are saying about how the displayed battery percentage is only an assessment. However I just downgraded to version 1.21.4 and the described behavior, where Sailing mode does not persists for discharge amounts less than 3%, does not occur. Sailing mode always persists consistently through a reboot, even if the discharge is only 1%. Also my testing consistently showed this cutoff of 3% occurring with version 1.22.2 without any charger being plugged in during the reboot sequence, as you mentioned that that could be a factor.

Therefore this does seem like it could potentially be a new limitation for Sailing mode introduced with version 1.22(.2). That is of course unless it is specifically related to something in my configuration and you or other users do not see the same in your testing. 

If, as you suggested, we should only expect Sailing mode to persist through reboots with thresholds above 3 to 5 percent and this behavior that I'm seeing can be reproduced on other machines, you should consider adding this as a new caveat to the Sailing mode feature explanation page as it does mention these low discharge amounts as its goal to prevent.

https://apphousekitchen.com/feature-explanation-sailing-mode/
> Since it is theoretically healthier for the battery to be charged 10% once instead of 1% 10 times, we have included the Sailing Mode feature.

In my use case, where at the end of the day I might be 1/2% away from my charge limit, this will introduce micro-charges that were previously prevented by Al Dente (before version 1.22). If due to new limitations this cannot be overcome I can understand and I'll work around it (let the battery drop to 67% before rebooting). 

Thank you again for all your efforts.
Freddy.

kvaellning, 2023-08-03:
> Hi @kvaellning,
> 
> It does not matter if you restart or turn off and on the MacBook manually. However, in order for all of this to work, it is important to unplug the MacBook while macOS and AlDente are still running and plug it in again when macOS and AlDente are running again.
> 
> Best, Matthias

Dear @MatthiasKerbl,
I think this could be the cause for my problem - inconsistency. I'll check it the next days with various scenarios (and really look at my steps :) ) and keep you updated if anything is off.

(Again, I don't really understand why the restrictions are so hard and not simply interval based, but you know why you are doing it this way).

Btw, I am running a MacBook Pro 14-inch 2021 with an Apple M1 Max, Ventura 13.5.

cheers,
Juliane

kvaellning, 2023-08-04:
Tried it again today:
- charge limit @40%, Difference 20%
- charged to upper limit, sailing mode active
- unplugged the MacBook & turned it off
- turned it on today, done some stuff, battery level @35%
- plugged it in - MacBook begins to charge

@MatthiasKerbl where can I find an older version? I just want to check if I also get this behaviour with an older AlDente Version.

cheers, Juliane

shahin365, 2023-08-04:
M1 Macbook Air, macOS 13.5. Version **1.22.2** It works for me without any issues.

Freddy500, 2023-08-04:
> where can I find an older version? I just want to check if I also get this behaviour with an older AlDente Version.

@kvaellning: You can find the dmg files on the Github releases page:

 https://github.com/davidwernhart/AlDente-Charge-Limiter/releases

This is the version before 1.22:
https://github.com/davidwernhart/AlDente-Charge-Limiter/releases/tag/1.21.4

kvaellning, 2023-08-04:
In 1.21.4, everything also works here.
Really strange issue, tbh.

With 1.22.2 it worked through 2 restarts. Unplugged it, turned it off for like 4 hours, restartet again (and waited for everything to load), now it starts again to load.

Cheers,
Juliane

Pi88no, 2023-08-13:
On my MacBook Pro (2021), Sailing Mode does not work even with 1.22.2. 

After a restart, the battery is charged although the battery level is within the sailing range.

DavidBoone, 2023-10-12:
Same issue here intermittently (1.22.3), though it occurs even without a reboot.  Usually I can unplug, work a bit, replug and it remains in Sail mode.  Occasionally however, when I plug in again it begins charging to my upper limit.  The frustrating thing is that there's no way to engage Sail mode manually, once it has decided to charge, there's no way to stop it without messing with the charge limit itself.

I can understand it may be tricky to determine whether the user intends to charge when plugging in, but a simple 3rd "Sail" button in addition to the existing Discharge & Top Up buttons would do the trick.

pavel-bogdashin, 2023-10-17:
The same for me. Sailing mode worked on 1.22.2 but after update on 1.22.3 it does not.

MacBook Pro (2021) M1 Pro
MacOS Ventura 13.6

myneur, 2023-11-11:
Same problem for me. Both on 1.22 and 1.23

Seems to be conditioned by how much the battery charge drops. Sailing 75-35 % works if I unplug and use the battery just for couple of percents, but when used for a longer period (e. g. when battery drops more than 10 %) it starts charging when plugged in despite being in the sailing range. 

I did not find any exact threshold when the Sailing stops to work. I don't know if it is the percentage or time on battery or something else. But it is consistent, that when I use the battery shorty it works vs. when I use it for longer period, it does not work. 

The sailing mode was the reason I bought it. I guess many might have the same reason for the purchase.  
@MatthiasKerbl is there any estimation when this will be fixed? Thanks

MatthiasKerbl, 2023-11-30:
Hi everyone,

We have just released AlDente 1.24 which includes a comprehensive overhaul of the Sailing Mode feature.
The Sailing Mode will now consistently persist unplugging, restarting, sleep, and closing/opening AlDente. Furthermore, the Sailing Mode will now not only start from the upper limit but also when you change the Charge Limit so that the current battery percentage ends up within the sailing range.

Best,
Matthias

closed 2023-11-30 by MatthiasKerbl

richard4911, 2025-02-23:
I too am using Sailing Mode set to 20% - 80% but my M3 MacBook Pro stays at 80% whilst plugged in to the charger when, I believe, it should discharge to 20% and then charge to 80% - or am I wrong in this assumption?

Any help/advice would be very welcome.

I'm using AlDente Pro V1.31.1

MatthiasKerbl, 2025-02-24:
> I too am using Sailing Mode set to 20% - 80% but my M3 MacBook Pro stays at 80% whilst plugged in to the charger when, I believe, it should discharge to 20% and then charge to 80% - or am I wrong in this assumption?
> 
> Any help/advice would be very welcome.
> 
> I'm using AlDente Pro V1.31.1

Hi @richard4911,

This is the intended behavior of the Sailing Mode. It is a common misconception that the Sailing Mode actively charges and discharges your MacBook between the sailing range. This would be much worse for the battery.
You can read a detailed description of what the Sailing Mode does on our blog "[Feature Explanation: Sailing Mode](https://apphousekitchen.com/feature-explanation-sailing-mode/)"

Best,
Matthias