AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #979: Crackling/hum when using external sound card

Yegor248, 2023-06-21:
Hello! 
I have peculiar problem when using Al dente pro on MacBook Pro m2. 
I’m a musician who performs life using external sound card. I’m using Presonus Quantum 2626 with thunderbolt 3 connection. Sound card can charge the MacBook slowly. 
When Al dente is blocking charge through sound card the output of the sound card is distorted (loud noise/crackling/hum). Sound is distorted in all apps that use external sound card. When Al dente is not blocking charge everything is normal. 
When I connect MagSafe charging (weather it is blocked or not by Al dente) everything is normal. Can sound distortion be fixed?

macOS Ventura 13.3.1 (22E261)
Al dente pro 1.21.4
MacBook Pro 14 M2Max
Presonus Quantum 2626

CMMChris, 2023-06-29:
This is a hardware problem related to the power circuitry inside your thunderbolt device and not caused by AlDente. Contact the manufacturer of the accessory for support.

Yegor248, 2023-06-29:
Haha, Presonus support told me to contact Al Dente support)

MatthiasKerbl, 2023-06-29:
Hi @Yegor248,

This is definitely the first time we are hearing about such an issue. Do you have the possibility to test your setup with another MacBook to figure out if it is maybe a MacBook hardware issue?

Best,
Matthias

Yegor248, 2023-06-30:
I can test with an old MacBook Pro 15 2015, would that be of any help?

MatthiasKerbl, 2023-08-22:
> I can test with an old MacBook Pro 15 2015, would that be of any help?

Hi @Yegor248,

did you test it with the MacBook Pro 15 2015?

Best,
Matthias

samasante, 2023-12-12:
I can confirm that this is also an issue when using AlDente to limit charge while plugged into an external audio interface (Focusrite Scarlett 4i4).

I can reproduce this issue consistently as seen in the attached GIF. When the limit is lowered (and the bar becomes blue) I immediately hear a loud feedback / ground loop noise coming from my speakers.

[image]

- AlDente Pro 1.2.4
- M1 Macbook Pro (Sonoma)

samasante, 2023-12-12:
Opened #1124 with additional information

MatthiasKerbl, 2026-10-06:
Hi everyone,

We have continued the discussion about this topic in #1124, which has already been closed. I will close this thread to keep the discussion in one place.

If you are still experiencing this issue with the latest version of AlDente, please let us know in a new issue and include a debug file.

Best,
Matthias

closed 2026-10-07 by MatthiasKerbl