AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #981: No longer working after restart

Pppau, 2023-06-24:
Al dente worked perfectly when I installed it, but then my mac restarted for an OS update, and now Aldente just doesn't work at all.

Not only does the Discharge feature no longer work, the charge limiter has no effect either.
I tried deleting it and reinstalling it, nothing changes.

I did accept the installation of the helper.

**MacBook (please complete the following information):**
 - MacBook Pro with M1 pro
 - macOS Ventura 13.4.1
 - AlDente Free
 - Version 1.22

Pppau, 2023-06-24:
Additional information: When I toggle Discharge on, the charging symbol on the system battery icon disappears (indicating that the battery is no longer charging) but it reappears after ~5 seconds.

Pppau, 2023-06-24:
I've used it on battery to get it under the limit, and now the charging LED alternates between orange (~10 seconds) and green (~3 seconds)
It still didn't stop the charging when it went over the limit

Pppau, 2023-06-26:
Edit: works again after another restart

MatthiasKerbl, 2023-06-29:
Hi @Pppau,

This might have been a known issue after macOS updates.
You can read more about it and how to fix it here #768.

Best,
Matthias