AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #984: When the pop-up is open, the menu bar auto-hides and moves the popup upwards

ADTC, 2023-06-28:
**Describe the bug**
This is a slightly unique situation, as most macOS users don't enable menu bar auto-hide. _But I do._

The problem with Al Dente is that if its popup is open, the menu bar hides when I move the mouse away, pulling the popup upwards. This doesn't happen with any other icon in the menu bar (neither macOS built-in icon or a third party app icon), in which case the menu bar stays visible as long as the icon's popup menu is open.

**To Reproduce**
Steps to reproduce the behavior:
1. In Desktop & Dock (in System Settings), set "Automatically hide and show the menu bar" to "Always".
2. Open the AlDente pop-up by clicking its icon in the menu bar.
3. Move the mouse pointer away and observe.

**Expected behavior**
When the pop-up is open, the menu bar should remain visible even if the mouse is moved away. When the mouse is clicked somewhere else, the pop-up should close and the menu bar should hide away.

**Screenshots**
Pop-up opened:

[image]

Mouse moved away:

[image]

(You can see the pop-up moved upwards in relation to the window below, and the arrow is cut off. Although not captured here, the top-left corner is hidden by the camera notch.)

**MacBook:**
 - MBP 16" 2021 M1 Pro
 - Ventura 13.4 (22F66)
 - AlDente Free
 - 1.22
 - Mostly default. Show on startup and Dock icon are disabled.

**Additional context**
I'm just guessing here, but it could be because you're using some sort of custom container implementation for the pop-up instead of the standard menu container provided by Apple. I think the standard container could possibly include your custom contents in it, based on what I see with Apple's own Control Center and Now Playing widget:

[image]

[image]

These don't hide the menu and move up when the mouse pointer is moved away.

MatthiasKerbl, 2023-06-29:
Hi @ADTC,

You are right, I am experiencing the same issue. We will look into it and try to fix it with one of the next updates.

Best,
Matthias

MatthiasKerbl, 2024-09-03:
Hi @ADTC,

We have just released AlDente 1.28.1 which includes a fix for this issue. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2024-09-03 by MatthiasKerbl