AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #985: Popup UI broken

fidwj01, 2023-06-29:
**Describe the bug**
The UI is broken when Power Flow is enabled in the menu bar popup

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Click on 'Stats'
3. Enable 'Show in Popover'
4. Click on the AlDente icon on the menu bar

**Expected behavior**
Proper UI appearance

**Screenshots*
[image]
*

**MacBook (please complete the following information):**
 - MacBook Pro (15-inch, 2017)
 - macOS 13.4 Ventura
 - AlDente Pro
 - 1.22
 - List of all enabled/disabled settings in AlDente: 'Check Automatically', 'Launch at Login', 'Stop charging when sleeping', 'Stop charging when powered off', 'Sailing Mode', 'Heat Protection', 'Show in Popover'

MatthiasKerbl, 2023-06-29:
Hi @fidwj01,

Thank you for letting us know about this bug. This might be an issue with this specific MacBook model. We will investigate this issue and let you know when there is a fix.

Best,
Matthias

TremotinoVolante, 2023-06-30:
Griaß di,

same problem and some more informations:

To Reproduce
Steps to reproduce the behavior:
not only in the popover mode but also in the stats window. The screenshots are taken from the popover view, but the visualisation is exaclty the same in the stats window.

Some issues with the bug:
- when the macbook is on battery power the wattage value flickers heavily. instaead when it is charing no flickering occur
- the icon of the macbook is placed over the settings button on the top right corner.

on battery (wattage value not displayed):
[image]

on battery:
[image]

charging:
[image]

sailing mode (without wattage value since flickering):
[image]

MacBook (please complete the following information):

MacBook Pro (15-inch, 2017)
macOS 13.4.1 Ventura
AlDente Pro
1.22
List of all enabled settings in AlDente: 
'Check Automatically', 'Launch at Login', 'Stop charging when sleeping', 'Stop charging when powered off', 'Sailing Mode', 'Heat Protection', 'Show in Popover', 'Hardware Battery Percentage' Everything else is disabled.

kcleong, 2023-07-04:
Wattage label is flickering both on battery and on AC power:
https://github.com/davidwernhart/AlDente-Charge-Limiter/assets/201650/9283c4d3-a01b-4602-9112-4ff4a858b517

MacBook Pro (15" 2017)
macOS 13.4.1 Ventura
AlDente Pro 1.22
List of all enabled settings in AlDente:
"Automatic discharge", "Disable sleep until charge limit", "Sailing mode:, "Heat protection", "Only al dente icon", "Show in popover"

MatthiasKerbl, 2023-07-19:
Hi everyone,

Good news, we were able to fix the issue and will release a new update soon.

Best,
Matthias

fidwj01, 2023-07-31:
@MatthiasKerbl I just updated to 1.22.2, I’m still seeing the same exact issue.

fidwj01, 2023-08-01:
> Hi @fidwj01 and @TremotinoVolante,
> 
> Thank you for reporting back and we are sorry it still does not work for you. For the development, we have used the 13" MacBook Pro 2017, and usually, it is similar to the 15" from 2017. However, this time it seems to be different. We will try to get our hands on the 15" version to try to get it to work on that one as well.
> 
> Best, Matthias
I appreciate that.
Just out of curiosity as I’m also in the software engineering industry. How can the UI behave differently when running on the same OS versions and same resolutions (I presume it’s working on 18, 19 MBP 15-inch models)?

fidwj01, 2023-08-01:
@MatthiasKerbl 
I appreciate that!
Just out of curiosity as I’m also in the software engineering industry. How can the UI behave differently when running on the same OS versions and same resolutions (I presume it’s working on 18, 19 MBP 15-inch models)?

MatthiasKerbl, 2023-08-01:
> @MatthiasKerbl I appreciate that! Just out of curiosity as I’m also in the software engineering industry. How can the UI behave differently when running on the same OS versions and same resolutions (I presume it’s working on 18, 19 MBP 15-inch models)?

Hi @fidwj01,

The issue seems to be caused by small hardware differences which are messing with the UI.

Best,
Matthias

MatthiasKerbl, 2023-10-10:
Hi everyone,

We have just released AlDente 1.22.3 which includes a fix for this issue. To get AlDente 1.22.3 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Cheers,
Matthias

TremotinoVolante, 2023-10-10:
> Hi everyone,
> 
> We have just released AlDente 1.22.3 which includes a fix for this issue. To get AlDente 1.22.3 you just need to update within the app under Settings/General/Check for Updates.
> 
> Please report back if it works for you.
> 
> Cheers, Matthias

It works! 

It works while using grid power, on battery and while charging the battery...

Thanks.

closed 2023-10-11 by MatthiasKerbl