AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #987: Sailing mode setting resets when setting charge limit to 20%

Proxymiity, 2023-07-02:
**Describe the bug**
Setting the charge limit to 20% resets the sailing mode parameter to 0%. The parameter is never restored when raising the charge limit back to, say, 80%.

**To Reproduce**
Steps to reproduce the behavior:
1. Set a charge limit of 80%
2. Set sailing difference of 5%
3. Set the charge limit to 20%
4. The sailing difference gets reset to 0%
5. Set the charge limit to 80% again
6. The sailing difference stays at 0%

**Expected behavior**
The sailing mode should be "temporarily disabled" instead of being reset to zero percent.

**MacBook (please complete the following information):**
 - MacBook Pro 18,2 (16in, 2021, M1 Max/64G/2T)
 - macOS Ventura 13.4 (22F66)
 - AlDente Pro 1.22
 - List of all enabled/disabled settings in AlDente:
 - Sailing Mode
 - Automatic Discharge
 - Disable Sleep until Charge Limit
 - Lock Screen when Closed
 - Hardware Battery Percentage
 - Show percentage
 - Show Battery: AlDente Status
 - Control Magsafe LED

MatthiasKerbl, 2023-07-19:
Hi @Proxymiity,

Thank you for your detailed bug report. We will fix this issue with the next update soon.

Best,
Matthias

MatthiasKerbl, 2023-11-29:
Hi @Proxymiity ,

We have just released AlDente 1.24 which includes a fix for this issue. To get AlDente 1.24 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2023-11-29 by MatthiasKerbl